Xilinx/finn #309: Cannot export the model to .onnx file with brevitas FINNManager.export

oskarnatan, 2021-04-22:
Hello,

Currently, I am developing a custom model for my own research. I would like to make my model able to inference faster when it is deployed on my xilinx PYNQ Z2 board. So far, I have read FINN project website and other resources and currently following the instruction from this page: https://finn.readthedocs.io/en/latest/end_to_end_flow.html

To be more specific, I am currently struggling with the model export step. The model is trained with brevitas and can be exported to .pth file successfully. However, the error arises when I export the model to .onnx file with FINNManager.export below:
```
FINNManager.export(model, export_path='model_weights_finn.onnx', input_t=in_tensor)
  File "/home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/base.py", line 57, in export
    return cls.export_onnx(module, input_shape, export_path, input_t, **kwargs)
  File "/home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/base.py", line 105, in export_onnx
    torch.onnx.export(module, input_t, export_path, **kwargs)
  File "/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/__init__.py", line 225, in export
    return utils.export(model, args, f, export_params, verbose, training,
  File "/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py", line 85, in export
    _export(model, args, f, export_params, verbose, training, input_names, output_names,
  File "/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py", line 647, in _export
    proto, export_map = graph._export_onnx(
RuntimeError: ONNX export failed: Couldn't export operator aten::upsample_bilinear2d
```
Followed with this information:
```
Defined at:
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/functional.py(3151): interpolate
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/modules/upsampling.py(141): forward
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/modules/module.py(709): _slow_forward
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/modules/module.py(725): _call_impl
/home/oskar/CARLA/CARLA_0.9.10.1/autonomous-vehicle/deeplearning/archq.py(1837): forward
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/modules/module.py(709): _slow_forward
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/modules/module.py(725): _call_impl
/home/oskar/.local/lib/python3.8/site-packages/torch/jit/_trace.py(116): wrapper
/home/oskar/.local/lib/python3.8/site-packages/torch/jit/_trace.py(125): forward
/home/oskar/.local/lib/python3.8/site-packages/torch/nn/modules/module.py(727): _call_impl
/home/oskar/.local/lib/python3.8/site-packages/torch/jit/_trace.py(1148): _get_trace_graph
/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py(342): _trace_and_get_graph_from_model
/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py(379): _create_jit_graph
/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py(409): _model_to_graph
/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py(632): _export
/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/utils.py(85): export
/home/oskar/.local/lib/python3.8/site-packages/torch/onnx/__init__.py(225): export
/home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/base.py(105): export_onnx
/home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/base.py(57): export
train2q.py(991): main
train2q.py(1053): <module>

Graph we tried to export:
graph(%0 : Float(1:327680, 20:16384, 128:128, 128:1, requires_grad=0, device=cpu)):
  %165 : Float(1:327680, 3:16384, 128:128, 128:1, requires_grad=0, device=cpu) = onnx::Slice[axes=[1], ends=[3], starts=[0]](%0) # /home/oskar/CARLA/CARLA_0.9.10.1/autonomous-vehicle/deeplearning/archq.py:1779:0
  %166 : Float(3:0, 255:1, requires_grad=0, device=cpu) = onnx::Constant[value=<Tensor>]() # /home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/finn/handler/act.py:144:0
  %167 : Tensor = onnx::MultiThreshold[domain="finn.custom_op.general", out_dtype="UINT8"](%165, %166)
  %168 : Float(requires_grad=0, device=cpu) = onnx::Constant[value={-128}]()
  %169 : FloatTensor = onnx::Add(%167, %168)
  %170 : Float(requires_grad=0, device=cpu) = onnx::Constant[value={0.00686275}]()
  %171 : Float(1:49152, 3:16384, 128:128, 128:1, requires_grad=0, device=cpu) = onnx::Mul(%169, %170) # /home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/finn/handler/act.py:144:0
  %172 : Double(16:27, 3:9, 3:3, 3:1, requires_grad=0, device=cuda:0) = onnx::Constant[value=<Tensor>]() # /home/oskar/.local/lib/python3.8/site-packages/brevitas/export/onnx/finn/handler/parameter.py:173:0
  %173 : Tensor = onnx::Conv[dilations=[1, 1], group=1, kernel_shape=[3, 3], pads=[1, 1, 1, 1], strides=[1, 1], weight_qnt="INT8"](%171, %172)
  %174 : Float(1:16, 16:1, 1:1, 1:1, requires_grad=0, device=cpu) = onnx::Constant[value=<Tensor>]()
  %175 : FloatTensor = onnx::Add(%173, %174)

the rest is omitted......
```

Yes, my model contains an upsampling layer and It seems that the export module has an issue with it. I have also tried the brevitas QuantUpSampling layer but I also got an issue.

Do you have any suggestions for me? I need to export my model to .onnx file so that I can continue to the next step described in https://finn.readthedocs.io/en/latest/end_to_end_flow.html

thank you in advance

maltanar, 2021-04-22:
Hi Oskar, in general, export issues should be reported in the Brevitas repository and not in the FINN compiler repo.  But I can add that the FINN compiler unfortunately does not support upsampling layers at the moment. You would have to add your own export and HLS conversion flow to FINN to support this, which is not yet documented.

oskarnatan, 2021-04-23:
I see. thanks

closed 2021-04-23 by oskarnatan