Xilinx/finn #312: InvalidGraph: [ONNXRuntimeError] : 10 : INVALID_GRAPH : This is an invalid model. Error in Node:Unsqueeze_3 : Node (Unsqueeze_3) has input size 1 not in range [min=2, max=2].

oskarnatan, 2021-04-23:
Hello,

I want to try this notebook tutorial https://github.com/Xilinx/finn/tree/master/notebooks
I have install brevitas:
`pip install brevitas`

and finn by git clone this repo and:
`pip install .`

and other necessary python packages. However, I got this error when executing this notebook (https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/bnn-pynq/tfc_end2end_example.ipynb)

```
---------------------------------------------------------------------------
InvalidGraph                              Traceback (most recent call last)
<ipython-input-24-84a83ad50a7e> in <module>
      5 
      6 model = model.transform(InferShapes())
----> 7 model = model.transform(FoldConstants())
      8 model = model.transform(GiveUniqueNodeNames())
      9 model = model.transform(GiveReadableTensorNames())

~/.local/lib/python3.8/site-packages/finn/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup, fix_float64)
    137         model_was_changed = True
    138         while model_was_changed:
--> 139             (transformed_model, model_was_changed) = transformation.apply(
    140                 transformed_model
    141             )

~/.local/lib/python3.8/site-packages/finn/transformation/fold_constants.py in apply(self, model)
     52                 # this node has no dynamic inputs, only constant ones -- so we can
     53                 # do constant folding.
---> 54                 oxe.execute_node(n, execution_context, graph)
     55                 # use the execution result as an initializer
     56                 model.set_initializer(node_out, execution_context[node_out])

~/.local/lib/python3.8/site-packages/finn/core/onnx_exec.py in execute_node(node, context, graph, return_full_exec_context)
    128                 input_dict[inp] = context[inp]
    129 
--> 130             sess = rt.InferenceSession(node_model.SerializeToString())
    131             output_list = sess.run(None, input_dict)
    132 

~/.local/lib/python3.8/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py in __init__(self, path_or_bytes, sess_options, providers, provider_options)
    278 
    279         try:
--> 280             self._create_inference_session(providers, provider_options)
    281         except RuntimeError:
    282             if self._enable_fallback:

~/.local/lib/python3.8/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py in _create_inference_session(self, providers, provider_options)
    307             sess = C.InferenceSession(session_options, self._model_path, True, self._read_config_from_model)
    308         else:
--> 309             sess = C.InferenceSession(session_options, self._model_bytes, False, self._read_config_from_model)
    310 
    311         # initialize the C++ InferenceSession

InvalidGraph: [ONNXRuntimeError] : 10 : INVALID_GRAPH : This is an invalid model. Error in Node:Unsqueeze_3 : Node (Unsqueeze_3) has input size 1 not in range [min=2, max=2].
```

I change nothing to the notebook. How to deal with this issue? thanks in advance

maltanar, 2021-04-26:
Hi Oskar. The only supported way to install FINN is in Docker at the moment. From the README:

> Getting Started
> Please see the Getting Started page for more information on requirements, installation, and how to run FINN in different modes. Due to the complex nature of the dependencies of the project, we only support Docker-based execution of the FINN compiler at this time.

FINN is sensitive to the versions of the packages it depends on, in this case the onnx package is probably the culprit. Please use the provided Docker infrastructure instead.

oskarnatan, 2021-04-26:
Hello,

thanks for the reply. I am sorry I didn't read it previously. Now I am able to run it in Docker.
However, I got this error when executing https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/cybersecurity/3-build-accelerator-with-finn.ipynb

[image]

Yes, I am sure that I didn't set the VIVADO_PATH previously. Could you tell me how to set the VIVADO_PATH? because I didn't see a clear explanation on this page https://finn.readthedocs.io/en/latest/getting_started.html

All I do is just:
`sh run-docker.sh`
from finn repo as instructed.

maltanar, 2021-04-26:
Can you try:
```
export VIVADO_PATH=/home/oskar/Xilinx/Vivado/2019.1
bash run-docker.sh
```

oskarnatan, 2021-04-26:
Shown this warning/error?
```
mv: cannot move '/workspace/finn/bdf/deprecated' to '/workspace/finn/board_files/deprecated': Directory not empty
mv: cannot move '/workspace/finn/bdf/microzed_7010' to '/workspace/finn/board_files/microzed_7010': Directory not empty
mv: cannot move '/workspace/finn/bdf/microzed_7020' to '/workspace/finn/board_files/microzed_7020': Directory not empty
mv: cannot move '/workspace/finn/bdf/minized' to '/workspace/finn/board_files/minized': Directory not empty
mv: cannot move '/workspace/finn/bdf/picozed_7010_fmc2' to '/workspace/finn/board_files/picozed_7010_fmc2': Directory not empty
mv: cannot move '/workspace/finn/bdf/picozed_7015_fmc2' to '/workspace/finn/board_files/picozed_7015_fmc2': Directory not empty
mv: cannot move '/workspace/finn/bdf/picozed_7020_fmc2' to '/workspace/finn/board_files/picozed_7020_fmc2': Directory not empty
mv: cannot move '/workspace/finn/bdf/picozed_7030_fmc2' to '/workspace/finn/board_files/picozed_7030_fmc2': Directory not empty
mv: cannot move '/workspace/finn/bdf/ultra96v2' to '/workspace/finn/board_files/ultra96v2': Directory not empty
mv: cannot move '/workspace/finn/bdf/ultrazed_3eg_iocc' to '/workspace/finn/board_files/ultrazed_3eg_iocc': Directory not empty
mv: cannot move '/workspace/finn/bdf/ultrazed_3eg_pciecc' to '/workspace/finn/board_files/ultrazed_3eg_pciecc': Directory not empty
mv: cannot move '/workspace/finn/bdf/ultrazed_7ev_cc' to '/workspace/finn/board_files/ultrazed_7ev_cc': Directory not empty
bash: /home/oskar/Xilinx/DocNav/.settings64-DocNav.sh: No such file or directory
bash: /home/oskar/Xilinx/SDK/2019.1/.settings64-SDK_Core_Tools.sh: No such file or directory
bash: /home/oskar/Xilinx/Model_Composer/2019.1/.settings64-Model_Composer.sh: No such file or directory
```
However, when I open the notebook again and execute all of them. It works!
So, what are these warnings/errors mean?

And also, I check only 1 core out of 12 cores is currently working. I read that the default worker is set to 1. Thus, if I want all cores to work, can I just do this? so that it will process the conversion quickly.
```
export NUM_DEFAULT_WORKERS=12
export VIVADO_PATH=/home/oskar/Xilinx/Vivado/2019.1
bash run-docker.sh
```

oskarnatan, 2021-04-27:
I have tried to set NUM_DEFAULT_WORKERS=6 in run-docker.sh but the working CPU core is still 1.

maltanar, 2021-04-27:
Regarding the "no such file or directory" errors for DocNav etc, you can ignore them: https://finn-dev.readthedocs.io/en/latest/faq.html#i-am-getting-docnav-and-model-composer-errors-when-launching-the-docker-image

For the multi-core transformations, setting the environment variable should be enough. You may want to verify that the environment variable is propagated correctly into the Docker container. Please be aware that not all transformations support parallel execution.

oskarnatan, 2021-04-28:
I see. thanks

closed 2021-04-28 by oskarnatan