Xilinx/finn #313: build_dataflow doesn't go past hls_ipgen step, it does work when specifying the folding myself

FrankHogervorst, 2021-04-27:
Converting my network graph gets stuck when using build_dataflow. It does work if I specify the folding myself and using build_custom.
Currently it got stuck in the StreamingFCLayerBatch7, previously it was failing at StreamingFCLayerBatch5. 

Output log of StreamingFCLayerBatch7:
```

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
  **** IP Build 2902112 on Wed May 27 22:43:36 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2020.1/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2020.1/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'ubuntu' on host 'finn_dev_ubuntu' (Linux_x86_64 version 5.4.0-1041-aws) on Tue Apr 27 13:33:15 +0000 2021
INFO: [HLS 200-10] In directory '/tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f'
Sourcing Tcl script '/tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f/hls_syn_StreamingFCLayer_Batch_7.tcl'
HLS project: project_StreamingFCLayer_Batch_7
HW source dir: /tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f
INFO: [HLS 200-10] Creating and opening project '/tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f/project_StreamingFCLayer_Batch_7'.
INFO: [HLS 200-10] Adding design file '/tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f/top_StreamingFCLayer_Batch_7.cpp' to the project
INFO: [HLS 200-10] Creating and opening solution '/tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f/project_StreamingFCLayer_Batch_7/sol1'.
INFO: [HLS 200-10] Setting target device to 'xcku060-ffva1517-1-c'
WARNING: [XFORM 203-506] Disable code size check when do loop unroll.
WARNING: [ANALYSIS 214-1] Skip long-run-time warning caused by lots of load/store instructions.
WARNING: [HLS 200-483] The 'config_rtl -auto_prefix' command is deprecated and will be removed in a future release. Use 'config_rtl -module_auto_prefix' as its replacement.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file '/tmp/finn_dev_ubuntu/code_gen_ipgen_StreamingFCLayer_Batch_7_oulfs75f/top_StreamingFCLayer_Batch_7.cpp' ... 
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:02:25 ; elapsed = 00:02:27 . Memory (MB): peak = 1532.988 ; gain = 1099.754 ; free physical = 9808 ; free virtual = 24491
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:02:25 ; elapsed = 00:02:27 . Memory (MB): peak = 1532.988 ; gain = 1099.754 ; free physical = 9808 ; free virtual = 24491
INFO: [HLS 200-10] Starting code transformations ...
INFO: [HLS 200-489] Unrolling loop 'Loop-3.1' (/workspace/finn-hlslib/mvau.hpp:138) in function 'void Matrix_Vector_Activate_Batch<512u, 512u, 4u, 1u, 1u, Slice<ap_int<2>, 2u>, Slice<ap_int<2>, 2u>, Identity, ap_uint<8>, ap_uint<2>, FixedPointWeights<4u, ap_int<2>, 1u, 65536u>, ThresholdsActivation<512u, 1u, 2u, ap_int<12>, ap_int<2>, -1, std::less_equal<ap_int<12> > >, ap_resource_lut>(hls::stream<FORWARD_REFERENCE, 0>&, hls::stream<FORWARD_REFERENCE, 0>&, FORWARD_REFERENCE const&, FORWARD_REFERENCE const&, int, FORWARD_REFERENCE const&)' completely with a factor of 1.
INFO: [HLS 200-489] Unrolling loop 'Loop-3.1.1' (/workspace/finn-hlslib/mvau.hpp:140) in function 'void Matrix_Vector_Activate_Batch<512u, 512u, 4u, 1u, 1u, Slice<ap_int<2>, 2u>, Slice<ap_int<2>, 2u>, Identity, ap_uint<8>, ap_uint<2>, FixedPointWeights<4u, ap_int<2>, 1u, 65536u>, ThresholdsActivation<512u, 1u, 2u, ap_int<12>, ap_int<2>, -1, std::less_equal<ap_int<12> > >, ap_resource_lut>(hls::stream<FORWARD_REFERENCE, 0>&, hls::stream<FORWARD_REFERENCE, 0>&, FORWARD_REFERENCE const&, FORWARD_REFERENCE const&, int, FORWARD_REFERENCE const&)' completely with a factor of 1.
INFO: [HLS 200-489] Unrolling loop 'Loop-3.2' (/workspace/finn-hlslib/mvau.hpp:149) in function 'void Matrix_Vector_Activate_Batch<512u, 512u, 4u, 1u, 1u, Slice<ap_int<2>, 2u>, Slice<ap_int<2>, 2u>, Identity, ap_uint<8>, ap_uint<2>, FixedPointWeights<4u, ap_int<2>, 1u, 65536u>, ThresholdsActivation<512u, 1u, 2u, ap_int<12>, ap_int<2>, -1, std::less_equal<ap_int<12> > >, ap_resource_lut>(hls::stream<FORWARD_REFERENCE, 0>&, hls::stream<FORWARD_REFERENCE, 0>&, FORWARD_REFERENCE const&, FORWARD_REFERENCE const&, int, FORWARD_REFERENCE const&)' completely with a factor of 1.
INFO: [HLS 200-489] Unrolling loop 'Loop-3.3' (/workspace/finn-hlslib/mvau.hpp:163) in function 'void Matrix_Vector_Activate_Batch<512u, 512u, 4u, 1u, 1u, Slice<ap_int<2>, 2u>, Slice<ap_int<2>, 2u>, Identity, ap_uint<8>, ap_uint<2>, FixedPointWeights<4u, ap_int<2>, 1u, 65536u>, ThresholdsActivation<512u, 1u, 2u, ap_int<12>, ap_int<2>, -1, std::less_equal<ap_int<12> > >, ap_resource_lut>(hls::stream<FORWARD_REFERENCE, 0>&, hls::stream<FORWARD_REFERENCE, 0>&, FORWARD_REFERENCE const&, FORWARD_REFERENCE const&, int, FORWARD_REFERENCE const&)' completely with a factor of 1.
INFO: [HLS 200-489] Unrolling loop 'Loop-3.3.1' (/workspace/finn-hlslib/mvau.hpp:165) in function 'void Matrix_Vector_Activate_Batch<512u, 512u, 4u, 1u, 1u, Slice<ap_int<2>, 2u>, Slice<ap_int<2>, 2u>, Identity, ap_uint<8>, ap_uint<2>, FixedPointWeights<4u, ap_int<2>, 1u, 65536u>, ThresholdsActivation<512u, 1u, 2u, ap_int<12>, ap_int<2>, -1, std::less_equal<ap_int<12> > >, ap_resource_lut>(hls::stream<FORWARD_REFERENCE, 0>&, hls::stream<FORWARD_REFERENCE, 0>&, FORWARD_REFERENCE const&, FORWARD_REFERENCE const&, int, FORWARD_REFERENCE const&)' completely with a factor of 1.

```

Files can be found here: https://we.tl/t-rMC5WT200w

This issue is a follow-up from the issue raised on Gitter on 5th of April 2021 by FrankHogervorst_gitlab

System:
Vivado 2020.1
Ubuntu 18.04
FINN commit af783db8dc2a1d2e95bd569d39464b935520b6d2

auphelia, 2023-02-14:
Hi @FrankHogervorst , I cannot see any error message in the output log of StreamingFCLayerBatch7. Could you check if your issue still persists when using the latest FINN release?

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-03-06 by auphelia