Xilinx/finn #321: Netron no longer works

mguzzina, 2021-05-06:
Today I downloaded the current version of the FINN repository, previously I believe I was using 0.4b.

Due to the [changes](https://github.com/lutzroeder/netron/commit/5762c05c9f65cfd2854587646fe9a6180e914503#diff-7a33ce62074239294a32a9e2a8aa39ba15b8c44dc2bacecc1552688d7ccb6a03) made to netron's server.py file the [showInNetron()](https://github.com/Xilinx/finn/blob/master/src/finn/util/visualization.py) function no longer works:
```
In [3]: showInNetron("./output_estimates_only/intermediate_models/1_step_tidy_up
   ...: .onnx")                                                                 
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-ddc552014674> in <module>
----> 1 showInNetron("./output_estimates_only/intermediate_models/1_step_tidy_up.onnx")

/workspace/finn/src/finn/util/visualization.py in showInNetron(model_filename)
     37 
     38 def showInNetron(model_filename):
---> 39     netron.start(model_filename, port=8081, host="0.0.0.0")
     40     return IFrame(src="http://0.0.0.0:8081/", width="100%", height=400)

TypeError: start() got an unexpected keyword argument 'port'
```

netron.start() now expects "address" as an argument, a tuple containing both the old host and port arguments.
The fix is theoretically trivial:
`netron.start(model_filename, port=8081, host="0.0.0.0")`
should become
`netron.start(model_filename, address=("0.0.0.0", 8080))`

If I try this however I'm left unable to connect to the Netron application via browser but that is likely an issue with Netron itself.

An old netron library version or [netron's own app](https://netron.app/) can be used as a workaround.

VanRillaerMike, 2021-05-06:
Hey, I've had the same problem.
This works for me:
[image]

This is also an interesting read, if you should (just like me) makes the mistake of putting 'localhost' or 127.0.0.1 in the address tuple. Instead of 0.0.0.0:
https://stackoverflow.com/questions/59179831/docker-app-server-ip-address-127-0-0-1-difference-of-0-0-0-0-ip#:~:text=0.0%20and%20it's%20like%20a,request%20to%20the%20current%20machine.

maltanar, 2021-05-06:
Hello, this has already been fixed in the `dev` branch, you may want to opt for that instead. This will go into `master` for the oncoming v0.6 release.

Relevant PR https://github.com/Xilinx/finn/pull/284

mguzzina, 2021-05-06:
Great to hear, I will definitely try the dev branch!

closed 2021-05-06 by mguzzina