Xilinx/finn #327: apply Floorplan error for 'model.find_consumer' when the consumer is AddStreams_Batch

simplelins, 2021-05-11:
please check!
the input of  'AddStreams_Batch' node are "StreamingDataWidthConverter_Batch_6_out0" and "StreamingDataWidthConverter_Batch_4_out0". when the node '
StreamingDataWidthConverter_Batch_4' find the consumer that return none, because the function find_consumer only select from input[0]
`    def find_consumer(self, tensor_name):
        """Finds and returns the node that consumes the tensor with given name.
        Currently only works for linear graphs."""
        all_inputs = [x.input[0] for x in self._model_proto.graph.node]
        try:
            consumer_ind = all_inputs.index(tensor_name)
            return self._model_proto.graph.node[consumer_ind]
        except ValueError:
            return None`
[image]
[image]

[image]
[image]
[image]

auphelia, 2023-02-14:
Hi @simplelins , could you please try if that issue still persists with the new FINN release?

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-03-06 by auphelia