Xilinx/finn #328: Question about the matching of finn and onnx versions

YZW-explorer, 2021-05-12:
Hello author, I am trying to implement this code https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/cybersecurity/2-export-to-finn-and-verify.ipynb
I want to ask what the version of FINN is, my .onnx file is exported using onnx version 1.8.1

maltanar, 2021-05-17:
Hello YZW-explorer, you should use the version of ONNX that your current version of the FINN docker image depends on, e.g. for the master branch:

https://github.com/Xilinx/finn/blob/master/requirements.txt#L8

`onnx==1.7.0`

In general my recommendation would be to do the ONNX export inside the FINN docker container to make sure the versions of Brevitas and ONNX are compatible with what the compiler expects.

YZW-explorer, 2021-05-17:
Thank you very much! I am trying to deploy my QNN model on the board following the steps here. https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/bnn-pynq/tfc_end2end_example.ipynb
But I ran into a problem that I couldn’t solve,my error is as follows：
onnxruntime.capi.onnxruntime_pybind11_state.InvalidGraph: [ONNXRuntimeError] : 10 : INVALID_GRAPH : This is an invalid model. Error in Node:Slice_0 : Node (Slice_0) has input size 1 not in range [min=3, max=5].
The node displayed in Netron is：
[image]

auphelia, 2023-02-14:
Hi @YZW-explorer , I will close this issue because since the last post, we update the onnx and onnxruntime versions multiple times. In the current FINN we use:
```
onnx==1.13.0
onnxruntime==1.11.1
```
Please be aware, that onnx nodes also come in different opset versions. In FINN we use opset version 11.

closed 2023-02-14 by auphelia