Xilinx/finn #329: Concatenation, Maxpool2D, and Upsample are causing error in ZynqBuild. "AssertionError: Must have 1 AXI lite interface on IODMA nodes"

oskarnatan, 2021-05-14:
Hello,
I open this issue based on our discussion in Xilinx-FINN Gitter Community. @fpjentzsch and @maltanar answer me and ask me to open this issue and upload my onnx files. Here is the original thread in Gitter:

> Currently, I am developing my own segmentation model that contains several brevitas (qnn) and torch (nn) layers as follows:
> 
> qnn.QuantConv2D
> qnn.QuantReLU
> nn.MaxPool2D
> nn.Upsample
> nn.BatchNorm2D
> nn.Dropout
> nn.Sigmoid
> torch.cat
> FYI, I develop the model gradually by increasing the layer. And also, the training process and FINN's ONNX export works well. Then, I follow this notebook (https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/bnn-pynq/cnv_end2end_example.ipynb) to process my model further. Everything is working smoothly until step 3. Then, I got an error on this step: 4.Hardware Generation --> ZynqBuild. It says "AssertionError: Must have 1 AXI lite interface on IODMA nodes". Here is the detail:
> 
> [image]
> 
> Does anyone have the same problem? Any help would be appreciated. Thank you.

After that, I also made a lot of variants to my model and do more experiments. Based on the experiment, I found that these layers are causing that error "AssertionError: Must have 1 AXI lite interface on IODMA nodes" in ZynqBuild process:
- torch.cat
- nn.MaxPool2D
- nn.Upsample

I have tried to change those layers with brevitas layers but still resulting the same error:
- qnn.QuantTensor.cat
- qnn.QuantMaxPool2D
- qnn.QuantUpsample

Here are my onnx files ([model_onnx.zip](https://github.com/Xilinx/finn/files/6478215/model_onnx.zip)) generated with this notebook (https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/bnn-pynq/cnv_end2end_example.ipynb) for each variant:

1. Just fully convolutional model (Conv+BN+ReLU --> Conv+BN+ReLU --> Dropout... --> Conv+Sigmoid) with different output of C (feature maps channel) and the same HxW = 32x32 from input to the final output layer. This model can be built smoothly with no error.
2. Similar to model 1 but with more Conv+BN+ReLU. This model can also be built smoothly with no error.
3. Similar to model 1 but with torch.cat to concat encoder's feature maps with decoder's feature maps. This arises the error "AssertionError: Must have 1 AXI lite interface on IODMA nodes" in ZynqBuild process.
4. Similar to model 1 but with MaxPool2D and Upsample to downgrade and upgrade the HxW size of the tensor. This is also causing the error.
5. Similar to model 1 but with torch.cat, MaxPool2D, and Upsample. Of course, this one is also causing the error.

To be noted, since I want to create a segmentation model, I skip these operations:
- model = model.transform(InsertTopK(k=1))
- model = model.transform(absorb.AbsorbScalarMulAddIntoTopK())
- model = model.transform(to_hls.InferLabelSelectLayer())

Based on the notebook's explanation, those operations are for a classification model where we provide the highest prediction probability. Thus, I think I don't need this. I want to create a good segmentation model so that I need those layers to worked. Do you have any suggestions regarding my issue? Any help would be appreciated. Thank you.

oskarnatan, 2021-05-16:
Update: I just add more conv-BN-ReLU layers to my model and it arises error: Exception: Synthesis failed, no bitfile found

[image]

What happens?

maltanar, 2021-05-17:
Just to link up here as well, I've replied to this on the Gitter channel: https://gitter.im/xilinx-finn/community?at=60a225d7d523ee3d3e397cf7

closed 2023-02-14 by auphelia