Xilinx/finn #333: StreamingMaxPool_Batch and Pool_Batch Estimation

SpontaneousDuck, 2021-05-21:
`StreamingMaxPool_Batch` and `Pool_Batch` are missing the estimation function implantations from `HLSCustomOp`. Is there an easy way to calculate the usage for those blocks? Thanks!

maltanar, 2021-05-30:
Pooling layers don't typically contribute much to the overall resource footprint for e.g. convnets, this is why they've been omitted so far. Of course it'd be great to have estimators for them as well.

Our normal approach to develop layer resource estimations functions is to run HLS synthesis across a number of different configurations for that layer (which config/s make sense depends a bit on layer specifics; e.g. input size, kernel size and PE come to mind), put everything into a spreadsheet, then run regression guided by some intuition about what variables matter the most to create an analytical model.

auphelia, 2023-02-14:
I am closing this issue due to inactivity. Please feel free to reopen or create a new issue if you have more questions!

closed 2023-02-14 by auphelia