Xilinx/finn #341: Launch a Build: PYNQ Bitfile and Driver Failed

JackeyZhang1001, 2021-06-03:
Hi all,
I'm trying the official demo to build a PYNQ bitfile.  I use the notebook _3-build-accelerator-with-finn.ipynb_ in /end2end_example/cybersecurity, and  execute the step of Launch a Build: PYNQ Bitfile and Driver. For the parameter _board_, if I use "Pynq-Z1", it will be executed successfully. If I change to "Ultra96", I get the following error: raise Exception("Synthesis failed, no bitfile found")
The code is as follow:
```
import finn.builder.build_dataflow as build
import finn.builder.build_dataflow_config as build_cfg

model_file = "cybsec-mlp-verified.onnx"

final_output_dir = "output_final"

cfg = build.DataflowBuildConfig(
    output_dir          = final_output_dir,
    target_fps          = 1000000,
    synth_clk_period_ns = 10.0,
    board               = "Ultra96", # Ultra96
    shell_flow_type     = build_cfg.ShellFlowType.VIVADO_ZYNQ,
    generate_outputs=[
        build_cfg.DataflowOutputType.BITFILE,
        build_cfg.DataflowOutputType.PYNQ_DRIVER,
        build_cfg.DataflowOutputType.DEPLOYMENT_PACKAGE,
    ]
)

build.build_dataflow_cfg(model_file, cfg)
```

The error is as follow:
```
Building dataflow accelerator from cybsec-mlp-verified.onnx
Intermediate outputs will be generated in /tmp/finn_dev_ubuntu
Final outputs will be generated in output_final
Build log is at output_final/build_dataflow.log
Running step: step_tidy_up [1/15]
Running step: step_streamline [2/15]
Running step: step_convert_to_hls [3/15]
Running step: step_create_dataflow_partition [4/15]
Running step: step_target_fps_parallelization [5/15]
Running step: step_apply_folding_config [6/15]
Running step: step_generate_estimate_reports [7/15]
Running step: step_hls_ipgen [8/15]
Running step: step_set_fifo_depths [9/15]
Running step: step_create_stitched_ip [10/15]
Running step: step_measure_rtlsim_performance [11/15]
Running step: step_make_pynq_driver [12/15]
Running step: step_out_of_context_synthesis [13/15]
Running step: step_synthesize_bitfile [14/15]
Traceback (most recent call last):
  File "/workspace/finn/src/finn/builder/build_dataflow.py", line 128, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/workspace/finn/src/finn/builder/build_dataflow_steps.py", line 486, in step_synthesize_bitfile
    ZynqBuild(cfg.board, cfg.synth_clk_period_ns, cfg.enable_hw_debug)
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 140, in transform
    transformed_model
  File "/workspace/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py", line 326, in apply
    MakeZYNQProject(self.platform, enable_debug=self.enable_debug)
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 140, in transform
    transformed_model
  File "/workspace/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py", line 258, in apply
    raise Exception("Synthesis failed, no bitfile found")
Exception: Synthesis failed, no bitfile found
> /workspace/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py(258)apply()
-> raise Exception("Synthesis failed, no bitfile found")
```
Does anyone have the same problem? Any help would be appreciated. Thank you.

HenniOVP, 2021-06-03:
Hi @JackeyZhang1001,

thank you for reporting the issue!

I was able to reproduce the issue on the current dev-branch. And it seems like the error is caused by the vendor definition being changed for the Ultra96v1 in a recent commit of the corresponding Avnet repository, see: https://github.com/Avnet/bdf/blob/2d49cfc25766f07792c0b314489f21fe916b639b/deprecated/ultra96v1/1.2/board.xml#L21 in combination with https://github.com/Avnet/bdf/commit/e6725f5667ab61e2ae2e7ff10761d9ee7521d18a

I've created a new branch to address this change: https://github.com/Xilinx/finn/tree/fix/341
An this branch the build now runs as expected on my system. If you have time can you check out the branch and test if this also solves the problem on your side?

@maltanar It seems that it might be useful to discuss, if FINN should check out a certain commit of the Avnet repository in the future to avoid such unexpected changes. Additionally, it seems that the Ultra96v1 has been moved to a deprecation folder, though I'm not sure what this means in this case, since Avnet gives no further description. Anyhow, it might be useful to switch to the Ultra96v2 board in the future.

JackeyZhang1001, 2021-06-04:
> Hi @JackeyZhang1001,
> 
> thank you for reporting the issue!
> 
> I was able to reproduce the issue on the current dev-branch. And it seems like the error is caused by the vendor definition being changed for the Ultra96v1 in a recent commit of the corresponding Avnet repository, see: https://github.com/Avnet/bdf/blob/2d49cfc25766f07792c0b314489f21fe916b639b/deprecated/ultra96v1/1.2/board.xml#L21 in combination with [Avnet/bdf@e6725f5](https://github.com/Avnet/bdf/commit/e6725f5667ab61e2ae2e7ff10761d9ee7521d18a)
> 
> I've created a new branch to address this change: https://github.com/Xilinx/finn/tree/fix/341
> On this branch the build now runs as expected on my system. If you have time can you check out the branch and test if this also solves the problem on your side?
> 
> @maltanar It seems that it might be useful to discuss, if FINN should check out a certain commit of the Avnet repository in the future to avoid such unexpected changes. Additionally, it seems that the Ultra96v1 has been moved to a deprecation folder, though I'm not sure what this means in this case, since Avnet gives no further description. Anyhow, it might be useful to switch to the Ultra96v2 board in the future.

I git clone the FINN project from GitHub again, and then run this step in notebook again. However, the same error still occurred.
As a beginner of FPGA, I don't know how to solve this error, so I decided to give you feedback again and wait for your reply. 
I should be grateful for your help.

HenniOVP, 2021-06-04:
Hm, okay. The issue will likely be fixed with the next release of FINN, that's version 0.6.
When that happens you can clone the project again and check if the bug is gone?

JackeyZhang1001, 2021-06-05:
Now that this problem has been solved, I have successfully exported Ultra96 bitfiles locally

JackeyZhang1001, 2021-06-05:
Now that this problem has been solved, I have successfully exported Ultra96 bitfiles locally. Thanks for your help.

HenniOVP, 2021-06-05:
Great to hear! Then I'll close the issue.

commit 525ef94f (2021-06-04): Merge pull request #342 from Xilinx/fix/341

Fixes #341
closed 2021-06-05 by HenniOVP