Xilinx/finn #344: 'Normalize' preprocessing + 'MoveScalarLinearPastInvariants' results in 'RuntimeError: Inferred shape and existing shape differ in rank: (4) vs (2)'

pete-lennart, 2021-06-08:
Hi, I am working on a 4-bit quantized neural network and try to insert a normalization preprocessing step into my network. For this I am executing
```python
model = ModelWrapper("trained.onnx")
model = model.transform(InferShapes())
model = model.transform(FoldConstants())
input_shape = model.get_tensor_shape(model.graph.input[0].name)
preproc = NormalizePreProc(0.1307, 0.3081, 1)
preproc_model = ModelWrapper(FINNManager.export(preproc, input_shape))
model = model.transform(MergeONNXModels(preproc_model))
model.set_tensor_datatype(model.graph.input[0].name, DataType.UINT8)
```
The trained model is generated by
```python
DROPOUT = 0.2

class Network(Module):
    def __init__(self):
        super(Network, self).__init__()
        self.inp = QuantIdentity(bit_width=4)
        self.ido = Dropout(p=DROPOUT)
        self.fc0 = QuantLinear(784, 64, bias=False, bit_width=4)
        self.bn0 = BatchNorm1d(64)
        self.id0 = QuantReLU(bit_width=4)
        self.do0 = Dropout(p=DROPOUT)
        self.fc1 = QuantLinear(64, 64, bias=False, bit_width=4)
        self.bn1 = BatchNorm1d(64)
        self.id1 = QuantReLU(bit_width=4)
        self.do1 = Dropout(p=DROPOUT)
        self.fc2 = QuantLinear(64, 64, bias=False, bit_width=4)
        self.bn2 = BatchNorm1d(64)
        self.id2 = QuantReLU(bit_width=4)
        self.do2 = Dropout(p=DROPOUT)
        self.out = QuantLinear(64, 10, bias=False, bit_width=4)

    def forward(self, x):
        x = x.view(x.shape[0], -1)
        x = self.inp(x)
        x = self.ido(x)
        x = self.fc0(x)
        x = self.bn0(x)
        x = self.id0(x)
        x = self.do0(x)
        x = self.fc1(x)
        x = self.bn1(x)
        x = self.id1(x)
        x = self.do1(x)
        x = self.fc2(x)
        x = self.bn2(x)
        x = self.id2(x)
        x = self.do2(x)
        x = self.out(x)
        return x

class NetworkWrapper(Module):
    def __init__(self):
        super(NetworkWrapper, self).__init__()
        self.mdl = Network()
        self.lsm = LogSoftmax(dim=1)

    def forward(self, x):
        x = self.mdl(x)
        x = self.lsm(x)
        return x

network = NetworkWrapper()
optimizer = Adam(network.parameters())
train(network, optimizer=optimizer)

FINNManager.export(network.mdl, input_shape=(1, 1, 28, 28), export_path='trained.onnx')
```
When I am executing `model = model.transform(MoveScalarLinearPastInvariants())` to move the normalization after the initial reshape node it fails with the following error message:
```
(op_type:Add, name:Sub_0): Inferred shape and existing shape differ in rank: (4) vs (2)
Traceback (most recent call last):
  File "/mnt/c/Users/peter/Git/fpga-ml-inference/src/mnist_fc_4bit/3_transform.py", line 34, in <module>
    model = model.transform(MoveScalarLinearPastInvariants())
  File "/home/peter/.local/lib/python3.6/site-packages/finn/core/modelwrapper.py", line 140, in transform
    transformed_model
  File "/workspace/finn/src/finn/transformation/streamline/reorder.py", line 642, in apply
    model = model.transform(InferShapes())
  File "/home/peter/.local/lib/python3.6/site-packages/finn/core/modelwrapper.py", line 140, in transform
    transformed_model
  File "/home/peter/.local/lib/python3.6/site-packages/finn/transformation/infer_shapes.py", line 88, in apply
    model = ModelWrapper(si.infer_shapes(model.model))
  File "/home/peter/anaconda3/envs/py36/lib/python3.6/site-packages/onnx/shape_inference.py", line 35, in infer_shapes
    inferred_model_str = C.infer_shapes(model_str, check_type)
RuntimeError: Inferred shape and existing shape differ in rank: (4) vs (2)
```
This seems to be an issue of the Add / Sub node of the normalization which corresponds to the subtraction of the mean. The related line of code can be found in `finn/src/finn/util/pytorch.py:42`:
```python
x = x - torch.tensor(self.mean, device=x.device).reshape(1, self.channels, 1, 1)
```
**If I change this line to simply `x = x - self.mean` everything works completely fine.**
Though, it seems like there is a reason this line is made more 'complex'.

HenniOVP, 2021-07-01:
Hi Peter,
thank you for reporting this issue!
It's good to know that you already found a workaround. Could you maybe send me the `trained.onnx` file, which you used here? This would make debugging easier and faster. Thanks in advance.
Hendrik

pete-lennart, 2021-09-04:
This seems to be as minimal as it can get:

 ```
from brevitas.export import FINNManager
from brevitas.nn import QuantIdentity
from finn.core.modelwrapper import ModelWrapper
from finn.transformation.fold_constants import FoldConstants
from finn.transformation.general import ConvertSubToAdd
from finn.transformation.infer_shapes import InferShapes
from finn.transformation.merge_onnx_models import MergeONNXModels
from finn.transformation.streamline.reorder import MoveScalarLinearPastInvariants
from finn.util.pytorch import Normalize
from torch.nn import Module

class Network(Module):
    def __init__(self):
        super(Network, self).__init__()
        self.inp = QuantIdentity()

    def forward(self, x):
        x = x.view(x.shape[0], -1)
        x = self.inp(x)
        return x

model = ModelWrapper(FINNManager.export(Network(), input_shape=(1, 1, 28, 28), export_path='trained.onnx'))

model = model.transform(InferShapes())
model = model.transform(FoldConstants())
input_shape = model.get_tensor_shape(model.graph.input[0].name)
preproc = Normalize(0.1307, 0.3081, 1)
preproc_model = ModelWrapper(FINNManager.export(preproc, input_shape))
model = model.transform(MergeONNXModels(preproc_model))
model = model.transform(ConvertSubToAdd())
model = model.transform(MoveScalarLinearPastInvariants())
```

Trained ONNX file here:
[trained.zip](https://github.com/Xilinx/finn/files/7110242/trained.zip)

commit 1c1557c7 (2021-11-04): Merge pull request #402 from Xilinx/fix/344

Fixes #344
closed 2021-11-05 by maltanar