Xilinx/finn #345: vivado_hls PATH and IPGenBuilder

dgm96, 2021-06-08:
Hello,
I have Vivado 2019.1 installed and Ubuntu18.04. When I try to run ./run-docker.sh notebook, and then the following code within the notebook:
    test_pynq_board = "Pynq-Z1"
    target_clk_ns = 10
    
    from finn.transformation.fpgadataflow.make_zynq_proj import ZynqBuild 
    model = ModelWrapper("model_name.onnx")
    model = model.transform(ZynqBuild(platform = test_pynq_board, period_ns = target_clk_ns))
    model.save("model_name_Zynq.onnx")

I get the following error: AssertionError: vivado_hls not found in PATH

When I try to echo $PATH from within the Jupyter notebook on Docker, vivado_hls is not in PATH, but it is in PATH if I echo $PATH from my terminal. I am not sure what to do.

Running the same code, without Docker gives me the following error:
ImportError: cannot import name 'IPGenBuilder' from 'finn.util.fpgadataflow'

I also checked fpgadataflow and IPGenBuilder is not there. So, I'm not sure what to do, as this step is not working for me either using Docker or my own system.

closed 2021-06-08 by dgm96

roderick1014, 2021-12-23:
Hi! It happened to me as well. Have you solved the problem?

caoimhelynch, 2022-03-01:
@dgm96 @roderick1014 this is happening to me also, did either of you find a solution?