Xilinx/finn #355: bug for weights  data width when the mem_mode == "external"

simplelins, 2021-06-28:
Hi, when I try to compile the vgg fc layer, the SIMD is 32 ,PE is 32 and "external" weight type is 8bits, the error occured as below:
```
ERROR: [CF2XD 83-2219] Port 'm_axis_0' on component 'idma0' has unsupported data width '8192' for direct connection
```
we know the data width should less then 512,but at line https://github.com/Xilinx/finn/src/finn/custom_op/fpgadataflow/streamingfclayer_batch.py#1076  , that set data wights to  'self.get_weightstream_width()'. so how to support external wight for lager SIMD and PE?

maltanar, 2021-08-06:
Hi @simplelins, I don't think there is an easy fix for this, unfortunately. You could manually add a data width converter at the stitched IP level to do the 512->8192 up-conversion but this will cause a bottleneck for that particular FC layer since its weight stream will not be delivered fast enough. My recommendation would be to use either smaller SIMD, PE and weight quantization for that layer.

auphelia, 2023-02-14:
I am closing this issue due to inactivity. Please feel free to reopen it or create a new issue, if you have more questions!

closed 2023-02-14 by auphelia