Xilinx/finn #368: Experimentation with unsupported board (Z-turn Board)

rpitonak, 2021-08-10:
Hello,

I have recently got working [End2End example](https://github.com/Xilinx/finn/blob/master/notebooks/end2end_example/bnn-pynq/tfc_end2end_example.ipynb) on an officially unsupported board - [Z-turn board](https://www.xilinx.com/products/boards-and-kits/1-571ww1.html). I would like to share some of the steps and possibly open a discussion for some enhancements in FINN.

1. The first thing that I had to workaround was `pynq_part_map`. Currently, there is no easy way to override this by custom arguments, for example, this cell:

  ```python
  from finn.transformation.fpgadataflow.make_zynq_proj import ZynqBuild
  model = ModelWrapper(build_dir+"/tfc_w1_a1_set_folding_factors.onnx")
  model = model.transform(ZynqBuild(platform = pynq_board, period_ns = target_clk_ns))
  ```

  have to be changed to something like this:

  ```python
  from finn.transformation.fpgadataflow.make_zynq_proj import ZynqBuild
  model = ModelWrapper(build_dir+"/tfc_w1_a1_set_folding_factors.onnx")
  z_build = ZynqBuild(platform = "Pynq-Z2", period_ns = target_clk_ns)
  z_build.platform = "Zturn"
  z_build.fpga_part = "xc7z020clg400-1"
  model = model.transform(z_build)
  ```

  which means to initialize the `ZynqBuild` object with one of the supported boards, to avoid `KeyError` in `pynq_part_map` and then change the properties manually. I think that there should be a possibility to initialize this object with custom `platform` + `fpga_part` arguments.

2. The same thing had to be done by changing the FINN source code at this place (e.g with hard-coded `fpga_part`):
https://github.com/Xilinx/finn/blob/master/src/finn/transformation/fpgadataflow/make_zynq_proj.py#L245

3. Add a new `elseif` branch to: https://github.com/Xilinx/finn/blob/master/src/finn/transformation/fpgadataflow/templates.py#L125-L128
looking something like this:

```tcl
} elseif {$BOARD == "Zturn"} {
    set ZYNQ_TYPE "zynq_7000"
    set_property board_part myir.com:mys-7z020:part0:2.1 [current_project]
}
```

If somebody is interested, you can find board files for the z-turn board in [this GitHub repository](https://github.com/q3k/zturn-stuff/tree/master/boards) and import them to Vivado.

With these changes, I was able to deploy the model on board and perform the inference.

I would be glad to hear your feedback on the proposed changes and I am ready to contribute with a PR afterwards.

iwTao, 2022-06-09:
Hi, 
thank you for your answer that how to run on unsupported board.It's very helpful to me, but i have a little problem that is about this  code `set_property board_part myir.com:mys-7z020:part0:2.1 [current_project]`. I have a custom board without any support files, so i didn't find a url such as `myir.com:mys-7z020:part0:2.1`. If i need write the board file myself, what should i do something to `set_property board_part`?

preusser, 2022-06-09:
Hi @iwTao,
the board setting is a convenience setting that implements the bulk import of the pin-out constraints of the board. If you don't have one available, you may just leave it out and provide your detailed interface definitions as part of your constraints set. Vivado includes a _Constraints Wizard_ that provides guidance in this process when opened on a _synthesized design_. For the purpose of re-use, you may eventually choose to bundle the matured constraints in a board file.

iwTao, 2022-06-09:
Hi @ preusser,
thanks for your reply. As you said, if i just leave it out. The board_part is undefined. Then i configured the `Constraints Wizard` and created `.XAS`  file by export in Vivado. I think the `.XAS`  file is includes `Constraints Wizard`. So what should i do from `.XAS`  file to TCL script?

closed 2022-06-09 by preusser