Xilinx/finn #579: Dataflow partition fails

jacopoabramo, 2022-03-25:
Greetings everyone,

I'm trying to port a Brevitas model for Iris classification into FPGA using FINN. The model I'm using is shown [here](https://github.com/Xilinx/brevitas/issues/392).

I'm using a custom notebook by taking as a reference the `tfc_end2end_example` in the repository. I'm able to come up with a reasonable model up the HLS conversion step. This is the model that I have so far:

[image]

When performing the dataflow partition step though, FINN throws the following exception:

```python
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-8-8e79fe16bf71> in <module>
      2 
      3 model = ModelWrapper(build_dir + model_name + "_hls_layers.onnx")
----> 4 parent_model = model.transform(CreateDataflowPartition())
      5 parent_model.save(build_dir + model_name + "_dataflow_parent.onnx")
      6 showInNetron(build_dir + model_name + "_dataflow_parent.onnx")

/workspace/finn-base/src/finn/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup, fix_float64)
    139         model_was_changed = True
    140         while model_was_changed:
--> 141             (transformed_model, model_was_changed) = transformation.apply(
    142                 transformed_model
    143             )

/workspace/finn/src/finn/transformation/fpgadataflow/create_dataflow_partition.py in apply(self, model)
     76 
     77         # first, use the generic partitioning functionality to split up the graph
---> 78         parent_model = model.transform(
     79             PartitionFromLambda(
     80                 partitioning=assign_partition_id, partition_dir=self.partition_model_dir

/workspace/finn-base/src/finn/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup, fix_float64)
    139         model_was_changed = True
    140         while model_was_changed:
--> 141             (transformed_model, model_was_changed) = transformation.apply(
    142                 transformed_model
    143             )

/workspace/finn-base/src/finn/transformation/create_generic_partitions.py in apply(self, model)
    122                 for node in to_check:
    123                     if node is not None:
--> 124                         assert (
    125                             self.partitioning(node) != partition_id
    126                         ), """cycle-free graph violated: partition depends on itself"""

AssertionError: cycle-free graph violated: partition depends on itself
```

I don't exactly understand what this means, so any input is appreciated. In case you need more information I'll happily provide them. Thank you.

My setup is:

- OS: Ubuntu 20.04.4 LTS (running on Windows 10 with WSL2)
- Python: 3.8.10
- FINN branch: main

fpjentzsch, 2022-04-07:
Hi, dataflow partitioning fails because not all nodes were converted to their respective HLS Custom Op in the prior steps. Specifically, the MatMul nodes should have been converted to StreamingFCLayer_Batch nodes.

Do you apply the `InferQuantizedStreamingFCLayer()` transformation as part of the HLS conversion?

jacopoabramo, 2022-04-11:
Hi @fpjentzsch , thank you for your reply. Yes, adding the transformation to the HLS conversion step did the trick. I'll close the issue.

jacopoabramo, 2022-07-11:
Greetings,

I'm reopening this issue for the following reason: I changed the FINN version (I checked out the "dev" branch, commit hash 281af25500cfbfb8e7abd23c956e4599851fdda8) and tried re-running my notebook to generate again the IP core for my model. I noticed that some of the functionalities were moved to the qonnx module but apart from that everything was the same. My problem is in the pre-HLS conversion step. The model I have is the following:

[image]

Afterwards I apply the following transformations:

```python
from qonnx.transformation.bipolar_to_xnor import ConvertBipolarMatMulToXnorPopcount
from finn.transformation.streamline.round_thresholds import RoundAndClipThresholds
from qonnx.transformation.infer_data_layouts import InferDataLayouts
from qonnx.transformation.general import RemoveUnusedTensors

chkpt_name = build_dir + model_name + pre_hls + ".onnx"
model = model.transform(ConvertBipolarMatMulToXnorPopcount())
model = model.transform(absorb.AbsorbAddIntoMultiThreshold())
model = model.transform(absorb.AbsorbMulIntoMultiThreshold())

# absorb final add-mul nodes into TopK
model = model.transform(absorb.AbsorbScalarMulAddIntoTopK())
model = model.transform(RoundAndClipThresholds())

# bit of tidy-up
model = model.transform(InferDataLayouts())
model = model.transform(RemoveUnusedTensors())

model.save(chkpt_name)
showInNetron(chkpt_name)
```

This produces the following model:

[image]

Then I perform the HLS conversion as follows:

```python
import finn.transformation.fpgadataflow.convert_to_hls_layers as to_hls

chkpt_name = build_dir + model_name + pre_hls + ".onnx"
model = ModelWrapper(chkpt_name)
model = model.transform(to_hls.InferBinaryMatrixVectorActivation(mem_mode=mem_mode))
# TopK to LabelSelect
model = model.transform(to_hls.InferLabelSelectLayer())
# input quantization (if any) to standalone thresholding
model = model.transform(to_hls.InferThresholdingLayer())

chkpt_name = build_dir + model_name + hls_layers + ".onnx"

model.save(chkpt_name)
showInNetron(chkpt_name)
```

[image]

At this point though I should not be seeing anymore the MatMul layers as these should have been collapsed within the MultiThreshold layers. I'm not exactly sure I understand what's going on. At any rate I afterwards try to create a dataflow partition as follows:

```python
from finn.transformation.fpgadataflow.create_dataflow_partition import CreateDataflowPartition

chkpt_name = build_dir + model_name + hls_layers + ".onnx"
model = ModelWrapper(chkpt_name)
parent_model = model.transform(CreateDataflowPartition())

chkpt_name = build_dir + model_name + dataflow_part + ".onnx"
parent_model.save(chkpt_name)
showInNetron(chkpt_name)
```

Which then fails:

```python
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-9-9b65553d6b9e> in <module>
      3 chkpt_name = build_dir + model_name + hls_layers + ".onnx"
      4 model = ModelWrapper(chkpt_name)
----> 5 parent_model = model.transform(CreateDataflowPartition())
      6 
      7 chkpt_name = build_dir + model_name + dataflow_part + ".onnx"

/home/jacopo/git/finn/deps/qonnx/src/qonnx/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup)
    138         model_was_changed = True
    139         while model_was_changed:
--> 140             (transformed_model, model_was_changed) = transformation.apply(transformed_model)
    141         if cleanup:
    142             transformed_model.cleanup()

/home/jacopo/git/finn/src/finn/transformation/fpgadataflow/create_dataflow_partition.py in apply(self, model)
     78 
     79         # first, use the generic partitioning functionality to split up the graph
---> 80         parent_model = model.transform(
     81             PartitionFromLambda(
     82                 partitioning=assign_partition_id, partition_dir=self.partition_model_dir

/home/jacopo/git/finn/deps/qonnx/src/qonnx/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup)
    138         model_was_changed = True
    139         while model_was_changed:
--> 140             (transformed_model, model_was_changed) = transformation.apply(transformed_model)
    141         if cleanup:
    142             transformed_model.cleanup()

/home/jacopo/git/finn/deps/qonnx/src/qonnx/transformation/create_generic_partitions.py in apply(self, model)
    116                 for node in to_check:
    117                     if node is not None:
--> 118                         assert (
    119                             self.partitioning(node) != partition_id
    120                         ), """cycle-free graph violated: partition depends on itself"""

AssertionError: cycle-free graph violated: partition depends on itself
```

auphelia, 2022-07-20:
Hi @jacopoabramo , I see that you are using the `InferBinaryMatrixVectorActivation()` function. Is your network binary? I had a brief look into the model you've sent in the beginning of the conversation and it looked for me like you're using 4-bit. Is that the same model you're still using? If yes, you can try replacing that function with [`InferQuantizedMatrixVectorActivation()`](https://github.com/Xilinx/finn/blob/abc500078692f7dec1f67aa7af4dead879eb1513/src/finn/transformation/fpgadataflow/convert_to_hls_layers.py#L716). You can now also use the current main branch to run your model.

jacopoabramo, 2022-07-21:
Hi @auphelia , thanks for the reply. That did the trick. Now I'm having an issue because apparently Vitis HLS is not installed (which makes sense since I only installed Vivado). I'll keep the issue open just in case I find more problems.

jacopoabramo, 2022-07-21:
Hello again,

I now have the following output after trying to build the model with Vitis HLS. The code for the building is shown below (target is PYNQ-Z2):

```python
from finn.util.basic import pynq_part_map
from finn.transformation.fpgadataflow.make_zynq_proj import ZynqBuild

# set the correct part map
fpga_part = pynq_part_map[target]
target_clk_ns = 10

chkpt_name = build_dir + model_name + folding + ".onnx"
model = ModelWrapper(chkpt_name)
model = model.transform(ZynqBuild(platform = target, period_ns = target_clk_ns))

chkpt_name = build_dir + model_name + post_synth + ".onnx"
model.save(chkpt_name)
showInNetron(chkpt_name)
```

This is what I get as traceback:

```python
---------------------------------------------------------------------------
RemoteTraceback                           Traceback (most recent call last)
RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "/home/jacopo/git/finn/src/finn/transformation/fpgadataflow/hlssynth_ip.py", line 69, in applyNodeLocal
    inst.ipgen_singlenode_code()
  File "/home/jacopo/git/finn/src/finn/custom_op/fpgadataflow/hlscustomop.py", line 339, in ipgen_singlenode_code
    assert os.path.isdir(
AssertionError: IPGen failed: /tmp/finn_dev_jacopo/code_gen_ipgen_StreamingDataflowPartition_0_IODMA_0_sryta8e_/project_StreamingDataflowPartition_0_IODMA_0/sol1/impl/ip not found. Check log under /tmp/finn_dev_jacopo/code_gen_ipgen_StreamingDataflowPartition_0_IODMA_0_sryta8e_
"""

The above exception was the direct cause of the following exception:

AssertionError                            Traceback (most recent call last)
<ipython-input-12-bfbbda2f9c4e> in <module>
      8 chkpt_name = build_dir + model_name + folding + ".onnx"
      9 model = ModelWrapper(chkpt_name)
---> 10 model = model.transform(ZynqBuild(platform = target, period_ns = target_clk_ns))
     11 
     12 chkpt_name = build_dir + model_name + post_synth + ".onnx"

/home/jacopo/git/finn/deps/qonnx/src/qonnx/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup)
    138         model_was_changed = True
    139         while model_was_changed:
--> 140             (transformed_model, model_was_changed) = transformation.apply(transformed_model)
    141         if cleanup:
    142             transformed_model.cleanup()

/home/jacopo/git/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py in apply(self, model)
    353                 PrepareIP(self.fpga_part, self.period_ns)
    354             )
--> 355             kernel_model = kernel_model.transform(HLSSynthIP())
    356             kernel_model = kernel_model.transform(
    357                 CreateStitchedIP(

/home/jacopo/git/finn/deps/qonnx/src/qonnx/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup)
    138         model_was_changed = True
    139         while model_was_changed:
--> 140             (transformed_model, model_was_changed) = transformation.apply(transformed_model)
    141         if cleanup:
    142             transformed_model.cleanup()

/home/jacopo/git/finn/deps/qonnx/src/qonnx/transformation/base.py in apply(self, model)
    103         # Execute transformation in parallel
    104         with mp.Pool(self._num_workers) as p:
--> 105             new_nodes_and_bool = p.map(self.applyNodeLocal, old_nodes, chunksize=1)
    106 
    107         # extract nodes and check if the transformation needs to run again

/opt/conda/lib/python3.8/multiprocessing/pool.py in map(self, func, iterable, chunksize)
    362         in a list that is returned.
    363         '''
--> 364         return self._map_async(func, iterable, mapstar, chunksize).get()
    365 
    366     def starmap(self, func, iterable, chunksize=None):

/opt/conda/lib/python3.8/multiprocessing/pool.py in get(self, timeout)
    769             return self._value
    770         else:
--> 771             raise self._value
    772 
    773     def _set(self, i, obj):

AssertionError: IPGen failed: /tmp/finn_dev_jacopo/code_gen_ipgen_StreamingDataflowPartition_0_IODMA_0_sryta8e_/project_StreamingDataflowPartition_0_IODMA_0/sol1/impl/ip not found. Check log under /tmp/finn_dev_jacopo/code_gen_ipgen_StreamingDataflowPartition_0_IODMA_0_sryta8e_
```

I installed Vitis version 2022.1

auphelia, 2022-07-21:
`WARNING: [HLS 207-5567] Invalid Directive: for current device, RAM_S2P + URAM is invalid combination for BIND_STORAGE's option 'type + impl' (/home/jacopo/git/finn/deps/finn-hlslib/slidingwindow.h:116:49)
`
From this warning it looks like URAM is selected. If I see it correctly, your target device is a Pynq-Z2, that board doesn't have URAM, you will need to select a different `ram_style` for that node.

jacopoabramo, 2022-07-21:
> `WARNING: [HLS 207-5567] Invalid Directive: for current device, RAM_S2P + URAM is invalid combination for BIND_STORAGE's option 'type + impl' [...]

Apparently the entire folding step I copied from the original source was incorrect; now I removed it and it seems that synthesis is running (hopefully at some point it will end). Where can I look in the documentation for updated informations on how to properly select the folding parameters?

auphelia, 2022-07-27:
You can find documentation about the folding factors here: https://github.com/Xilinx/finn/blob/github-pages/docs/finn-sheduling-and-folding.pptx

Some of the FINN transformations (e.g. HLSSynthIP) can be parallelized using the env var `NUM_DEFAULT_WORKERS` (https://finn.readthedocs.io/en/latest/getting_started.html#environment-variables), the default is 4

jacopoabramo, 2022-09-16:
Closing issue as it is now solved.

closed 2022-09-16 by jacopoabramo