Xilinx/finn #582: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/finn_dev_nick/vivado_stitch_proj_kbkt75vx/all_verilog_srcs.txt'

guptamrinal, 2022-03-31:
This error arises while running end2end_example/cybersecurity/3-build-accelerator-with-finn.ipynb.

During 10th command in the notebook following step is not getting completed. 
Running step: step_set_fifo_depths [11/17]

Following error is shown:
Traceback (most recent call last):
  File "/workspace/finn/src/finn/builder/build_dataflow.py", line 166, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/workspace/finn/src/finn/builder/build_dataflow_steps.py", line 450, in step_set_fifo_depths
    model = model.transform(
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 141, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/set_fifo_depths.py", line 306, in apply
    sim = pyverilate_stitched_ip(model)
  File "/workspace/finn-base/src/finn/util/pyverilator.py", line 167, in pyverilate_stitched_ip
    with open(vivado_stitch_proj_dir + "/all_verilog_srcs.txt", "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/finn_dev_nick/vivado_stitch_proj_kbkt75vx/all_verilog_srcs.txt'

I've checked the above directory and the file "all_verilog_srcs.txt" is in fact not present there.Even if I create the file ,what should be it's content bcoz finn is trying to read something from that file.

fpjentzsch, 2022-04-07:
Hi, this might be the same issue as here: https://github.com/Xilinx/finn/discussions/497

Did you apply the Y2K22 patch to Vivado? (https://support.xilinx.com/s/article/76960?language=en_US)

auphelia, 2022-07-20:
Hi, just checking in to ask if your issue was solved with @fpjentzsch answer? Otherwise I would close the issue

closed 2022-07-27 by auphelia

guptamrinal, 2022-07-27:
Yes it got solved