Xilinx/finn #587: How to import custom model in tfc_end2end_verification

B-Willems, 2022-04-08:
### Discussed in https://github.com/Xilinx/finn/discussions/585

<div type='discussions-op-text'>

<sup>Originally posted by **B-Willems** April  6, 2022</sup>
I want to use the tfc_end2end_verification notebook to verify my finn model but am encountering a few issues.

In this notebook in the second cell, a model gets imported with the command get_test_model_trained("TFC", 1, 1). How can i import my own brevitas export/quantized model in this notebook?

Secondly in the verification of the example network using PyVerilator another weird thing happens. When trying to evaluate, the golden_output is an array of 10 values (as expected), but the output of the rtl simulation is 2. If I look closely at these values, the softmax of golden_output is also 2, so the prediction was successful, however the format in which these are given is not the same. Are there any known fixes for this issue?

An image of the executed cell with added prints can be found below.

Thanks in advance.

[image]</div>

auphelia, 2023-02-14:
Hi @B-Willems , if you have your model in the form of an .onnx model file, then you can simply importing it using ModelWrapper in FINN.
Otherwise, have a look at the basics notebook about Brevitas FINN-ONNX export (https://github.com/Xilinx/finn/blob/main/notebooks/basics/1_brevitas_network_import.ipynb), section 2 and 3. Or in one of our test cases: https://github.com/Xilinx/finn/tree/main/tests/brevitas

The error with global out should be solved by now, please let me know if your issue is solved!

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-03-06 by auphelia