Xilinx/finn #598: Hardware generation works in main branch with Vivado and Vivado HLS 2020.1 but no in freature/vitis_hls or feature/vitis_hls_2021.2

haoxi1999, 2022-05-11:
## Quick summary
Hi. I've been unable to generate hardware for my neural network model using FINN branches for VItis HLS (feature/vitis_hls or feature/vitis_hls_2021.2 with Vitis/Vivado tools 2021.2). While my project can be generated with main branch and Vivado tools 2020.1, attempts of generating the hardware using vitis_hls branch failed at the step of CreateStitchedIP when executing the ZynqBuild flow. 
## Details

Looking at the generated Vivado and Vitis HLS projects, I have noticed the vivado_stitch_proj_xxxxxxxx project fails at the step of setting property of the out_V port of the block design. Inspecting the StreamingDataflowPartition IP generated by Vitis HLS, I noticed that there's no out_V port from the IP block generated by Vitis HLS. Here's the difference of the IP blocks generated by Vivado HLS and Vitis HLS shown side by side.

Vivado HLS:
[image]
Vitis HLS
[image]

### Possible fix

The tcl file in the generated Vivado stiching project may need to be updated in for the Vitis HLS IP blocks.

auphelia, 2022-07-20:
Hi @haoxi1999 , sorry for the late reply. Can you try using the new FINN release for your project and check if it is working now?

auphelia, 2023-02-14:
I am closing this issue due to inactivity. 
Please reopen or create a new issue if you are facing more problems. Also please note that we have KV260 support with the newest FINN release, I hope this solves your issue!

closed 2023-02-14 by auphelia