Xilinx/finn #659: driver.py doesn't work on my Alveo U50

Benjamin-Laf, 2022-08-09:
Hi,

I am using FINN to generate some kernels but I have some problems when I launch the bitfile with the XRT driver on my Alveo U50.
Previously, I installed the partition on my card with the good version indicated on your site (xilinx.com) in comparaison to my Vitis tools version.

[image]

When I launched this command with the driver.py and the bitfile which corresponded at the project.

[image]

This command run but it doesn't finish for no reason. If I stop it, I obtain this message:

[image]

It's very annoying for me because I don't have the possibility to test my kernel on the card and I don't find the problem for the moment.

Thanks for your reply.

auphelia, 2023-02-14:
Hi @Benjamin-Laf , does your issue still persist?
If yes, it looks like the driver waits for the output DMA to have data. Did all the previous verification steps in your network generation work? Which FINN version are you using and which Vitis version?

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-03-06 by auphelia