Xilinx/finn #662: Error ZynqBuild- AttributeError: 'NoneType' object has no attribute 's'

student180529, 2022-08-10:
Hi,
after creating Onnx files with neural network from Brevitas I tried to transform it on ZynqBoard following to tfc_end2end_example.ipynb chapter 3. Hardware Build. Unfortunately this line: model  = model.transform(ZynqBuild... caused an error: 

 File "/mnt/c/Users/ptumiali/Downloads/finn/src/finn/transformation/fpgadataflow/insert_iodma.py", line 103, in <genexpr>
    get_by_name(x.attribute, "backend").s.decode("UTF-8") == "fpgadataflow"
AttributeError: 'NoneType' object has no attribute 's'

What should I do to fix this issue?

auphelia, 2023-02-14:
Hi @pawel-tumialis , are you still facing this issue? It looks like one of your layers is not an HLS layer, so doesn't have the attribute backend set to a value.

Could you try running your example with the new FINN release?

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-03-06 by auphelia