Xilinx/finn #701: QUICKTEST ALWAYS FAILS!!!

mlsearc, 2022-10-23:
I have defined Xİlinx path by using the methodology below.
`export FINN_XILINX_PATH=/tools/Xilinx`
My xilinx version is 2022.1 and I have also defined the Vitis Path too by the same method. As I go with the echo I am successful to get the data path.

After setting the enviromental variables my FINN always fails on the quick test with thousands of warnings, and  1 failed test. I really wonder FINN always fail, I am using Ubuntu 18.04 and I am pretty bored by not being able to use FINN.

I am using the latest FINN version which is updated 27 days ago of this post and its version is FINN v0.8.1 

I am able to open jupyter notebooks but I am not able to do the designs. Especially when it is the vivado ip generation phase. To inform you better I have pasted my log on a txt file. 

The failed status is 
=========================== short test summary info ============================

> FAILED tests/fpgadataflow/test_fpgadataflow_checksum.py::test_fpgadataflow_checksum

[finn_quicktest_log.odt](https://github.com/Xilinx/finn/files/9846537/finn_quicktest_log.odt)

mlsearc, 2022-10-23:
As another note here is how quicktest looks like:

> = 1 failed, 1248 passed, 259 skipped, 1 xpassed, 1014 warnings in 1509.79s (0:25:09) =

mlsearc, 2022-10-23:
Here is my error and warnings summary with the log that error mentions.

> __________________________ test_fpgadataflow_checksum __________________________
> [gw0] linux -- Python 3.8.5 /opt/conda/bin/python
> multiprocessing.pool.RemoteTraceback: 
> """
> Traceback (most recent call last):
>   File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 125, in worker
>     result = (True, func(*args, **kwds))
>   File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 48, in mapstar
>     return list(map(*args))
>   File "/home/best/Desktop/upd_finn/finn/src/finn/transformation/fpgadataflow/hlssynth_ip.py", line 69, in applyNodeLocal
>     inst.ipgen_singlenode_code()
>   File "/home/best/Desktop/upd_finn/finn/src/finn/custom_op/fpgadataflow/hlscustomop.py", line 339, in ipgen_singlenode_code
>     assert os.path.isdir(
> AssertionError: IPGen failed: /tmp/finn_dev_best/code_gen_ipgen_MatrixVectorActivation_0_bmf618_t/project_MatrixVectorActivation_0/sol1/impl/ip not found. Check log under /tmp/finn_dev_best/code_gen_ipgen_MatrixVectorActivation_0_bmf618_t
> """
> 
> The above exception was the direct cause of the following exception:
> 
>     @pytest.mark.fpgadataflow
>     def test_fpgadataflow_checksum():
>         # use a graph consisting of two fc layers to test
>         # checksum node insertion
>         model = create_two_fc_model()
>     
>         # set checksum output hook
>         for n in model.graph.node:
>             n0 = getCustomOp(n)
>             n0.set_nodeattr("output_hook", "checksum")
>     
>         model = model.transform(InsertHook())
>         model = model.transform(GiveUniqueNodeNames())
>         model = model.transform(GiveReadableTensorNames())
>         model = model.transform(InferShapes())
>     
>         assert (
>             len(model.get_nodes_by_op_type("CheckSum")) == 2
>         ), """Insertion of
>             checksum layers was unsuccessful"""
>     
>         # to verify the functionality of the checksum layer
>         # cppsim and rtlsim will be compared
>     
>         x = gen_finn_dt_tensor(DataType["INT32"], (1, 4))
>     
>         # cppsim
>         model = model.transform(SetExecMode("cppsim"))
>         model = model.transform(PrepareCppSim())
>         model = model.transform(CompileCppSim())
>         inp = {"global_in": x}
>         y_cppsim = oxe.execute_onnx(model, inp, return_full_exec_context=True)
>         checksum0_cppsim = y_cppsim["CheckSum_0_out1"]
>         checksum1_cppsim = y_cppsim["CheckSum_1_out1"]
>     
>         # in this test case scenario the checksums are equal
>         assert checksum0_cppsim == checksum1_cppsim, "CheckSums are not equal"
>     
>         # rtlsim
>         model = model.transform(InsertFIFO(True))
>         model = model.transform(GiveUniqueNodeNames())
>         model = model.transform(PrepareIP(test_fpga_part, target_clk_ns))
> >       model = model.transform(HLSSynthIP())
> 
> tests/fpgadataflow/test_fpgadataflow_checksum.py:178: 
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
> deps/qonnx/src/qonnx/core/modelwrapper.py:140: in transform
>     (transformed_model, model_was_changed) = transformation.apply(transformed_model)
> deps/qonnx/src/qonnx/transformation/base.py:105: in apply
>     new_nodes_and_bool = p.map(self.applyNodeLocal, old_nodes, chunksize=1)
> /opt/conda/lib/python3.8/multiprocessing/pool.py:364: in map
>     return self._map_async(func, iterable, mapstar, chunksize).get()
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
> 
> self = <multiprocessing.pool.MapResult object at 0x7f565c0651c0>, timeout = None
> 
>     def get(self, timeout=None):
>         self.wait(timeout)
>         if not self.ready():
>             raise TimeoutError
>         if self._success:
>             return self._value
>         else:
> >           raise self._value
> E           AssertionError: IPGen failed: /tmp/finn_dev_best/code_gen_ipgen_MatrixVectorActivation_0_bmf618_t/project_MatrixVectorActivation_0/sol1/impl/ip not found. Check log under /tmp/finn_dev_best/code_gen_ipgen_MatrixVectorActivation_0_bmf618_t
> 
> /opt/conda/lib/python3.8/multiprocessing/pool.py:771: AssertionError
> ----------------------------- Captured stderr call -----------------------------
> /tools/Xilinx/Vivado/2022.1/bin/loader: line 312: 23649 Killed                  "$RDI_PROG" "$@"
> 
[vitis_hls.log](https://github.com/Xilinx/finn/files/9846551/vitis_hls.log)

mlsearc, 2022-10-23:
Plus, as I go with the cnv flow of the the jupyter notebook on firefox browser, or chrome, I always get the not responding error on the hardware generation step. My PC stucks for a long time.

auphelia, 2022-10-24:
Hi @mlsearc , in the quicktest.sh, the number of pytest workers is set to auto in [line 3](https://github.com/Xilinx/finn/blob/main/docker/quicktest.sh#L3) (`: ${PYTEST_PARALLEL=auto}`). It might be that the processes kill each other if the workload is too high. Could you try setting this parameter to a fixed number?

auphelia, 2023-02-14:
I am closing this issue due to inactivity. 
Please feel free to reopen or create a new issue if the error persists!

closed 2023-02-14 by auphelia