Xilinx/finn #706: Unable to run a container from maltanar/finn:dev_latest

aneesullah, 2022-10-30:
Hi, 

The following command to start a container from prebuilt images

docker run --name au_finn -it maltanar/finn:dev_latest bash

Gives the following errors :

ERROR: /deps/qonnx is not a valid editable requirement. It should either be a path to a local project or a VCS URL (beginning with svn+, git+, hg+, or bzr+).
ERROR: /deps/finn-experimental is not a valid editable requirement. It should either be a path to a local project or a VCS URL (beginning with svn+, git+, hg+, or bzr+).
ERROR: /deps/brevitas is not a valid editable requirement. It should either be a path to a local project or a VCS URL (beginning with svn+, git+, hg+, or bzr+).
ERROR: /deps/pyverilator is not a valid editable requirement. It should either be a path to a local project or a VCS URL (beginning with svn+, git+, hg+, or bzr+).
ERROR: Unable to find FINN source code in
ERROR: Ensure you have passed -v <path-to-finn-repo>:<path-to-finn-repo> to the docker run command

Regards,
Anees

auphelia, 2023-02-14:
Hi @aneesullah, to use the prebuilt image please follow these steps: https://finn.readthedocs.io/en/latest/getting_started.html#using-a-prebuilt-image 
If you have set the env vars, you can simply use the ./run-docker.sh and it will use the prebuilt image.

But please be aware that we most likely deprecate the prebuilt Docker image support soon: https://github.com/Xilinx/finn/discussions/764

Ba1tu3han, 2023-02-26:
> 

Hello,

I encountered the same problem and I follow your reply. I have set the env variables and run "./run-docker.sh quicktest" and it is still giving the same error. 

Screenshots are below.

What is that I missed? I can not figure it out.

Thank you,

[image]
[image]
[image]

shahzaib3120, 2023-08-09:
go to the parent directory of finn and change permissions

<code>sudo chmod -R 777 finn/ </code>