Xilinx/finn #713: Exception: Output shapes disagree after node execution: found (3,) vs expected (0,)

mllearner98, 2022-11-22:
On my FINN execution I always get the exact same error. I could not figure it out about 3 months and this is the reason why I need a huge help please. 

I have model that is quantized and works perfectly fine with brevitas. So I was able to extract the quantized .pth extended model and It has a pretty good accuracy. As I try to import this model to the .onnx I was also successful.  

As I try to run the code below, in order to start the sequence for FINN, I always get the same error, which is also shown below.

```
import onnx
from finn.util.test import get_test_model_trained
import brevitas.onnx as bo
from qonnx.core.modelwrapper import ModelWrapper
from qonnx.transformation.infer_shapes import InferShapes
from qonnx.transformation.fold_constants import FoldConstants
from qonnx.transformation.general import GiveReadableTensorNames, GiveUniqueNodeNames, RemoveStaticGraphInputs

model = ModelWrapper("./mymodel.onnx")
model = model.transform(InferShapes())
model = model.transform(FoldConstants())
model = model.transform(GiveUniqueNodeNames())
model = model.transform(GiveReadableTensorNames())
model = model.transform(RemoveStaticGraphInputs())
model.save(build_dir + "/mymodel_tidy.onnx")
```

The error message:

> Exception                                 Traceback (most recent call last)
> <ipython-input-2-f35aaeb6b49f> in <module>
>       9 model = ModelWrapper("./pointnet_oneout_wsoft_inp.onnx")
>      10 model = model.transform(InferShapes())
> ---> 11 model = model.transform(FoldConstants())
>      12 model = model.transform(GiveUniqueNodeNames())
>      13 model = model.transform(GiveReadableTensorNames())
> 
> /home/best/Desktop/upd_finn/finn/deps/qonnx/src/qonnx/core/modelwrapper.py in transform(self, transformation, make_deepcopy, cleanup)
>     138         model_was_changed = True
>     139         while model_was_changed:
> --> 140             (transformed_model, model_was_changed) = transformation.apply(transformed_model)
>     141         if cleanup:
>     142             transformed_model.cleanup()
> 
> /home/best/Desktop/upd_finn/finn/deps/qonnx/src/qonnx/transformation/fold_constants.py in apply(self, model)
>      57                 # this node has no dynamic inputs, only constant ones -- so we can
>      58                 # do constant folding.
> ---> 59                 oxe.execute_node(n, execution_context, graph)
>      60                 # use the execution result as an initializer
>      61                 model.set_initializer(node_out, execution_context[node_out])
> 
> /home/best/Desktop/upd_finn/finn/deps/qonnx/src/qonnx/core/onnx_exec.py in execute_node(node, context, graph, return_full_exec_context, opset_version)
>      85             # use that index to index output_list
>      86             if output_list[list_ind].shape != context[outp].shape:
> ---> 87                 raise Exception(
>      88                     """Output shapes disagree after node execution:
>      89                     found %s vs expected %s"""
> 
> Exception: Output shapes disagree after node execution:
>                     found (3,) vs expected (0,)
> 

In order to give a further information, here is the information that I get by my model.

from qonnx.core.datatype import DataType

```
finnonnx_in_tensor_name = model_for_sim.graph.input[0].name
finnonnx_out_tensor_name = model_for_sim.graph.output[0].name
print("Input tensor name: %s" % finnonnx_in_tensor_name)
print("Output tensor name: %s" % finnonnx_out_tensor_name)
finnonnx_model_in_shape = model_for_sim.get_tensor_shape(finnonnx_in_tensor_name)
finnonnx_model_out_shape = model_for_sim.get_tensor_shape(finnonnx_out_tensor_name)
print("Input tensor shape: %s" % str(finnonnx_model_in_shape))
print("Output tensor shape: %s" % str(finnonnx_model_out_shape))
finnonnx_model_in_dt = model_for_sim.get_tensor_datatype(finnonnx_in_tensor_name)
finnonnx_model_out_dt = model_for_sim.get_tensor_datatype(finnonnx_out_tensor_name)
print("Input tensor datatype: %s" % str(finnonnx_model_in_dt.name))
print("Output tensor datatype: %s" % str(finnonnx_model_out_dt.name))
print("List of node operator types in the graph: ")
print([x.op_type for x in model_for_sim.graph.node])
```

> Input tensor name: inp.1
> Output tensor name: 359
> Input tensor shape: [1, 3, 1024]
> Output tensor shape: [1, 10]
> Input tensor datatype: FLOAT32
> Output tensor datatype: FLOAT32
> List of node operator types in the graph: 
> ['Shape', 'Gather', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'MaxPool', 'Flatten', 'MatMul', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'MatMul', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Unsqueeze', 'Concat', 'Unsqueeze', 'Concat', 'Shape', 'ConstantOfShape', 'Expand', 'Tile', 'MatMul', 'Mul', 'Add', 'Reshape', 'Add', 'Transpose', 'MatMul', 'Transpose', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Shape', 'Gather', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'MaxPool', 'Flatten', 'MatMul', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'MatMul', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Unsqueeze', 'Concat', 'Unsqueeze', 'Concat', 'Shape', 'ConstantOfShape', 'Expand', 'Tile', 'MatMul', 'Mul', 'Add', 'Reshape', 'Add', 'Transpose', 'MatMul', 'Transpose', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'Conv', 'Mul', 'Add', 'BatchNormalization', 'MaxPool', 'Flatten', 'MatMul', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'MatMul', 'Mul', 'Add', 'BatchNormalization', 'MultiThreshold', 'Mul', 'MatMul', 'Mul', 'Add', 'LogSoftmax']
> 

Please help to me, if any further information is needed or if you think that you can generate the bitstream and pynq driver for my model please write below your email.

auphelia, 2023-02-14:
Hi @mllearner98, could you try if your problem is solved with the latest release? 
Otherwise, the model files (in .onnx) before you get the error would be helpful for further investigation.

auphelia, 2023-03-06:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-03-06 by auphelia