Xilinx/finn #720: AttributeError: module 'numpy' has no attribute 'str' during step_hls_codegen [9/17]

rstar900, 2022-12-21:
File "..../finn/src/finn/util/data_packing.py", line 268, in elem2str
    if type(x) == str or type(x) == np.str_ or type(x) == np.str :
  File "/opt/conda/lib/python3.8/site-packages/numpy/__init__.py", line 284, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'str'

I get this error at step_hls_codegen [9/17].

Tried removing str as well, but did not solve the issue. The code is the same in dev branch.

maltanar, 2022-12-21:
Can you do a `pip freeze | grep numpy` and share the version of the numpy package running in the FINN Docker container? If it's not the same as what https://github.com/Xilinx/finn/blob/dev/requirements.txt#L7 asks for, this might indicate some other dependency has forced a newer version that breaks something. If this is the case, a workaround would be to`pip install numpy==1.22.0` in the Docker container before launching the FINN builds.

rstar900, 2022-12-21:
Yes, It turned out to be 1.24.0 in the docker. Had to run pip uninstall and then install. Just installing alone would not work. Thanks, it solved the issue, but now I face another issue. 

AssertionError: DWC OutWidth is bigger than InWidth and is not divisible by it.
                Please adjust PE and SIMD values so that OutWidth % InWidth = 0
                or alternatively use impl_style = vivado

You can have a look at the updated code https://github.com/rstar900/ML_And_Neural_Nets/blob/main/FINN_Lenet5_HLS_Code_Gen.ipynb

closed 2022-12-21 by rstar900

JJJWG, 2023-01-22:
hi,

how exactly did you changed the numpy dependency in docker? 

When I launch `./run-docker.sh notebook` I can't change the version with pip in the terminal. 

Otherwise when I launch `./run-docker.sh` I am able to change the version to 1.22.0 as mentioned above but don't know how to launch the notebooks from there. Also when I exit docker the version changes to numpy revert back to 1.24.0

I'm still pretty new to the concepts of docker and am confused about the differences between `./run-docker.sh notebook` and `./run-docker.sh`.

john

seniorgoogles, 2023-01-23:
> hi,
>
> how exactly did you changed the numpy dependency in docker?
>
> When I launch `./run-docker.sh notebook` I can't change the version with pip in the terminal.
>
> Otherwise when I launch [...]

Hi John, 

I guess that the Docker container is being recreated. You can start the Docker container with `./run-docker.sh notebook` and then call it with `docker exec -it CONTAINER_ID bash`. You can get the CONTAINER_ID by running the `docker ps` command in your terminal. Then just install numpy with the command `pip3 install numpy==1.22.0`. 

It worked for me.

Michael

rstar900, 2023-01-23:
> hi,
>
> how exactly did you changed the numpy dependency in docker?
>
> When I launch `./run-docker.sh notebook` I can't change the version with pip in the terminal.
>
> Otherwise when I launch [...]

Hi John,

As I mentioned before closing the issue, have a look at https://github.com/rstar900/ML_And_Neural_Nets/blob/main/FINN_Lenet5_HLS_Code_Gen.ipynb

I did the uninstall (really important even if it throws error), then install older version before going for the last step in the code. This is the only solution I found to be working in my case. So yes, these commands have to be put in the notebook so they are executed everytime you execute the notebook.

JJJWG, 2023-01-24:
Hi,

thank you very much guys!

I tried it with the uninstall and install in the notebook and it worked.

John

anguyen9630, 2023-02-11:
> > hi,
> > how exactly did you changed the numpy dependency in docker?
> > When I launch `./run-docker.sh notebook` I can't change the version with pip in the terminal.
> > Otherwise when I launch `./run-docker.sh` I am able to change the version to 1.22.0 as mentioned above but don't know how to launch the notebooks from there. Also when I exit docker the version changes to numpy revert back to 1.24.0
> > I'm still pretty new to the concepts of docker and am confused about the differences between `./run-docker.sh notebook` and `./run-docker.sh`.
> > john
> 
> Hi John,
> 
> As I mentioned before closing the issue, have a look at https://github.com/rstar900/ML_And_Neural_Nets/blob/main/FINN_Lenet5_HLS_Code_Gen.ipynb
> 
> I did the uninstall (really important even if it throws error), then install older version before going for the last step in the code. This is the only solution I found to be working in my case. So yes, these commands have to be put in the notebook so they are executed everytime you execute the notebook.
> 
> Rachit

`AssertionError: DWC OutWidth is bigger than InWidth and is not divisible by it.`: This one I do not know how to fix the issue it self, because some of the modules you cannot set PE and some you cannot set SIMD so technically it is never achievable. **HOWEVER, the solution is to instead use the Vivado config as suggested.** Use `force_rtl_conv_inp_gen  = True` and `rtlsim_use_vivado_comps = True` in the build config and then change your **ENSURE THAT YOUR SIMD COUNT IS <= 512**

anguyen9630, 2023-02-11:
> Yes, It turned out to be 1.24.0 in the docker. Had to run pip uninstall and then install. Just installing alone would not work. Thanks, it solved the issue, but now I face another issue. [...]

`AssertionError: DWC OutWidth is bigger than InWidth and is not divisible by it.`: This one I do not know how to fix the issue it self, because some of the modules you cannot set PE and some you cannot set SIMD so technically it is never achievable. **HOWEVER, the solution is to instead use the Vivado config as suggested.** Use `force_rtl_conv_inp_gen  = True` and `rtlsim_use_vivado_comps = True` in the build config and then change your **ENSURE THAT YOUR SIMD COUNT IS <= 512**