Xilinx/finn #733: AssertionError: IPGen failed (Running step: step_hls_ipgen [10/17])

Wintermute607, 2023-01-11:
I've come across this assertion error when trying to run the end2end examples:

```
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "/home/christian/Downloads/finn/src/finn/transformation/fpgadataflow/hlssynth_ip.py", line 73, in applyNodeLocal
    inst.ipgen_singlenode_code()
  File "/home/christian/Downloads/finn/src/finn/custom_op/fpgadataflow/hlscustomop.py", line 358, in ipgen_singlenode_code
    assert os.path.isdir(
AssertionError: IPGen failed: /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_x3s_r9op/project_MatrixVectorActivation_2/sol1/impl/ip not found. Check log under /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_x3s_r9op
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/christian/Downloads/finn/src/finn/builder/build_dataflow.py", line 168, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/home/christian/Downloads/finn/src/finn/builder/build_dataflow_steps.py", line 492, in step_hls_ipgen
    model = model.transform(HLSSynthIP())
  File "/home/christian/Downloads/finn/deps/qonnx/src/qonnx/core/modelwrapper.py", line 140, in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
  File "/home/christian/Downloads/finn/deps/qonnx/src/qonnx/transformation/base.py", line 111, in apply
    new_nodes_and_bool = p.map(self.applyNodeLocal, old_nodes, chunksize=1)
  File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 364, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/opt/conda/lib/python3.8/multiprocessing/pool.py", line 771, in get
    raise self._value
AssertionError: IPGen failed: /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_x3s_r9op/project_MatrixVectorActivation_2/sol1/impl/ip not found. Check log under /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_x3s_r9op

```

```
> /opt/conda/lib/python3.8/multiprocessing/pool.py(771)get()
    769             return self._value
    770         else:
--> 771             raise self._value
    772 
    773     def _set(self, i, obj):
```

Unfortunately, I am at a loss here.

Wintermute607, 2023-01-11:
Here's the vitis_hls.log. Money quote:
```
INFO: [HLS 200-1510] Running: config_rtl -deadlock_detection none
ERROR: [HLS 200-101] 'config_rtl': Unknown option '-deadlock_detection'.
ERROR: [HLS 200-101] 'config_rtl': Unknown option 'none'.
```
I wonder what could be the cause of this error. I am using Vivado v2021.2 (ML Edition, 64-bit) and I have also  (most likely) specified the paths correctly:
`source /opt/Xilinx/Vivado/2021.2/settings64.sh`
`source /opt/Xilinx/Vitis_HLS/2021.2/settings64.sh`
`export FINN_XILINX_PATH=/opt/Xilinx`
`export FINN_XILINX_VERSION=2021.2`
`export VIVADO_PATH=/tools/Xilinx/Vivado/2021.2`
`export VITIS_PATH=/opt/Xilinx/Vitis_HLS/2021.2`
`export PATH="/tools/Xilinx/Vivado/2021.2/bin:$PATH"`

Maybe theres something else going on.

**Complete logfile:**
```

****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2021.2 (64-bit)
  **** SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
  **** IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vitis_HLS/2021.2/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vitis_HLS/2021.2/bin/unwrapped/lnx64.o/vitis_hls'
/opt/Xilinx/Vitis_HLS/2021.2/tps/tcl/tcl8.5/tzdata/Europe/Dublin can't be opened.
INFO: [HLS 200-10] For user 'christian' on host 'finn_dev_christian' (Linux_x86_64 version 5.15.0-57-generic) on Wed Jan 11 16:21:53 +0000 2023
INFO: [HLS 200-10] On os Ubuntu 18.04.5 LTS
INFO: [HLS 200-10] In directory '/tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp'
Sourcing Tcl script '/tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/hls_syn_MatrixVectorActivation_2.tcl'
HLS project: project_MatrixVectorActivation_2
HW source dir: /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp
finn-hlslib dir: /home/christian/Downloads/finn/deps/finn-hlslib
custom HLS dir: /home/christian/Downloads/finn/custom_hls
INFO: [HLS 200-1510] Running: open_project project_MatrixVectorActivation_2
INFO: [HLS 200-10] Creating and opening project '/tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/project_MatrixVectorActivation_2'.
INFO: [HLS 200-1510] Running: add_files /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/top_MatrixVectorActivation_2.cpp -cflags -std=c++14 -I/home/christian/Downloads/finn/deps/finn-hlslib -I/home/christian/Downloads/finn/custom_hls
INFO: [HLS 200-10] Adding design file '/tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/top_MatrixVectorActivation_2.cpp' to the project
INFO: [HLS 200-1510] Running: set_top MatrixVectorActivation_2
INFO: [HLS 200-1510] Running: open_solution sol1
INFO: [HLS 200-10] Creating and opening solution '/tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/project_MatrixVectorActivation_2/sol1'.
INFO: [HLS 200-1505] Using default flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part xc7z020clg400-1
INFO: [HLS 200-1611] Setting target device to 'xc7z020-clg400-1'
INFO: [HLS 200-1510] Running: config_compile -disable_unroll_code_size_check -pipeline_style flp
WARNING: [XFORM 203-506] Disable code size check when do loop unroll.
WARNING: [HLS 200-643] The 'config_compile -disable_unroll_code_size_check' hidden command is deprecated and will be removed in a future release.
INFO: [HLS 200-1510] Running: config_interface -m_axi_addr64
INFO: [HLS 200-1510] Running: config_rtl -module_auto_prefix
INFO: [HLS 200-1510] Running: config_rtl -deadlock_detection none
ERROR: [HLS 200-101] 'config_rtl': Unknown option '-deadlock_detection'.
ERROR: [HLS 200-101] 'config_rtl': Unknown option 'none'.
SYNTAX
  config_rtl [OPTIONS]
    -header <string>
    -kernel_profile[=false|*true*]
    -module_auto_prefix[=true|*false*]
    -module_prefix <string>
    -mult_keep_attribute[=false|*true*]
    -register_all_io[=false|*true*]
    -register_reset_num <int:*0*>
    -reset <none | state | all | *control*>
    -reset_async[=false|*true*]
    -reset_level <low | *high*>

SEE ALSO
  www.xilinx.com/cgi-bin/docs/rdoc?v=2021.2;t=vitis+doc;d=vyw1583260160301.html

command 'config_rtl' returned error code
    while executing
"config_rtl -deadlock_detection none"
    (file "/tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/hls_syn_MatrixVectorActivation_2.tcl" line 25)
    invoked from within
"source /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/hls_syn_MatrixVectorActivation_2.tcl"
    invoked from within
"hls::main /tmp/finn_dev_christian/code_gen_ipgen_MatrixVectorActivation_2_wrqt9vpp/hls_syn_MatrixVectorActivation_2.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$newargs"
    (procedure "hls_proc" line 16)
    invoked from within
"hls_proc [info nameofexecutable] $argv"
INFO: [HLS 200-112] Total CPU user time: 5.42 seconds. Total CPU system time: 1 seconds. Total elapsed time: 6.8 seconds; peak allocated memory: 1.160 GB.
INFO: [Common 17-206] Exiting vitis_hls at Wed Jan 11 16:21:59 2023...

```

auphelia, 2023-01-11:
Hi @Wintermute607, could you upgrade your Vivado/Vitis installation? The `main` and the `dev` branch run with version 2022.1

Wintermute607, 2023-01-12:
Hello @auphelia, with **Vivado 2022.1**, the error no longer occurs. Thanks again for the quick help.

For anyone who is interested, here is the setup that makes the FINN compiler work as intended for me:

- Host Operating System: Ubuntu 22.04
- Docker version 20.10.22
- Xilinx Vivado 2022.1
- FINN Compiler as of January 12, 2023

closed 2023-01-12 by Wintermute607