Xilinx/finn #743: Quicktest Docker read-only Error

Ba1tu3han, 2023-01-30:
Hello,

I'm trying to install FINN but I get this error about docker which are shown in pics. I changed ownership of folders but it is still the same. I could not change ownership of "xinstall".  What should I do ? I haven't run any FINN code yet. 

Additionally, When I run "0_how_to_work_with_onnx", I get "ModuleNotFoundError: No module named 'finn.util.visualization'" error.

I'm following the instructions from getting started page. I have done first 3 steps. My Ubuntu version is 22.04. Is it about that? FINN version is V0.8.1 and Vivado version is 2022.1.

[image]
[image]

Bests,
Batuhan,

anguyen9630, 2023-02-07:
Hey, I don't know if it is too late but you should add yourself into the docker group so that you do not have to run docker with sudo if you haven't done it already: https://docs.docker.com/engine/install/linux-postinstall/

Otherwise, what fixed it for me, when I had to restart due to some issues with jupyter notebook, is to basically clear docker. 
Documentation for pruning: [https://docs.docker.com/config/pruning/](url)

`docker system prune`

Please be mindful that everything will be wiped, so if you have something else with docker you should back it up.

auphelia, 2023-02-14:
Thanks a lot @anguyen9630 for answering to the issue! @Ba1tu3han did that solve your problem? I will leave this issue open for a few days but then will close it as answered.

auphelia, 2023-02-20:
I am closing this due to inactivity. Please feel free to reopen or create a new issue if the problem persists!

closed 2023-02-20 by auphelia