Xilinx/finn #796: Obtained different result after calling an IP several times, seems like the data did not work properly inside.

KokyoK, 2023-04-08:
Hello, here we are facing a weird problem.
What we have done: 
- Created an .onnx model file,
- Created an IP with no warnings.
- Ran the C simulation and RTL simulation, every test passed.

What we are currently doing:
Trying to use the IP on Pynq-z2 board. But when we call an IP several times, it outputs different results.  Like this:
[image]
As you can see, the inputs of this IP are exactly the same. But the outputs are similar but different every time, and the first output is exactly correct, but others are not. Seems like it uses some internal data from the previous input.  However, if we reinitialize the Overlay every time we want to use it, it works properly but much slower, as shown in the following pic:
[image]

These are the onnx file and final test notebook we used. 
[ipynb&onnx.zip](https://github.com/Xilinx/finn/files/11182879/ipynb.onnx.zip)

auphelia, 2023-04-13:
Hi @KokyoK , could you please provide more information?
- Which FINN version and which tool versions (Vivado/Vitis) are you using?
- Is that one of our example networks or one you created/trained yourself?

KokyoK, 2023-04-13:
Hello @auphelia , 

- the versions are:
      Pynq: v2.6.1
      FINN: the newest version (from `main` branch)
      FINN-examples: v0.0.5
      Vivado: 2022.1

- it is the network we created and trained ourselves

Thanks

auphelia, 2023-04-20:
Hi @KokyoK,
While we're looking from our side if we can reproduce your issue, could you please update the PYNQ version to use 3.0.1 and see if the error persists?

KokyoK, 2023-04-24:
Hi @auphelia ,
I have tried to use PYNQ 3.0.1 and the same issue still occurs.

fionnodonohoe-xlnx, 2023-05-04:
Hi @KokyoK,
I am unable to build the bitstream from the provided ONNX file, could you please provide the original trained model?
Would it be possible to also share the 'input_permute.npy' file that is used by the notebook? Thanks

KokyoK, 2023-05-08:
Hi @fionnodonohoe-xlnx , here are the files:
`model.py` includes the model structure, which is a normal convolutional model. `weight.pt` is the trained weights loaded by the model. Also `input_permute.npy` is provided. 

Since we've build a bitstream with the provided ONNX, I guess there might be something wrong when we built bitstream.

[model.zip](https://github.com/Xilinx/finn/files/11417242/model.zip)

Thanks for your effort!

fionnodonohoe-xlnx, 2023-05-17:
Hi @KokyoK,

I tried creating the ONNX file from `model.py`. When adding `model.save(is_onnx=1)` after `model.eval()` I get the following error:
`RuntimeError: Given groups=1, weight of size [16, 40, 1, 3], expected input[16, 30, 1, 101] to have 40 channels, but got 30 channels instead`

I then changed the expected input to have 40 channels instead of 30 - only to get this error:
`RuntimeError: input_shape.size() > 0 || reshape.size() > 0INTERNAL ASSERT FAILED at "../torch/csrc/jit/passes/onnx/shape_type_inference.cpp":448, please report a bug to PyTorch. Reshape node should have at least one input size > 0 when constant folding.`

Are you also seeing this error? Maybe you could send on your `TCResNet8.onnx` file created from your script. I can try to put that ONNX file through the bitstream generation stage then.

KokyoK, 2023-05-22:
Hello @fionnodonohoe-xlnx ,
1. The correct way to save `.onnx` file is putting the following line in the main function:
```
    model = QuantizedTCResNet8(1, 40, 10)
    model.load("weight.pt")
    model.eval()

    import brevitas.onnx as bo
    export_onnx_path = "8b_weight_act_bias_net.onnx"
    input_shape = (1, 40, 1, 101)
    bo.export_finn_onnx(model, input_shape, export_onnx_path)
```
and run the main function.  The onnx file should be the same as I uploaded before. Sorry for the previous confusion. 

2. We built the bitstream with this file modified `build_dataflow_steps.py` basically we add line 327, and commented line 338. The file is attached.

[build_dataflow_steps.py.zip](https://github.com/Xilinx/finn/files/11527156/build_dataflow_steps.py.zip)

KokyoK, 2023-05-23:
Hi @fionnodonohoe-xlnx ,
You can simply remove all code related to TLAST in the code. Remove this line:
` model = model.transform(InsertTLastMarker(both=True))`
As this line is not related to this issue. We've tried to remove it and the issue still occurs.

fionnodonohoe-xlnx, 2023-05-24:
Hi @KokyoK ,
I went ahead and removed the TLastMarker insertion point. The bitstream failed to generate this time due to a lack of resources available for the Pynq FPGA part. I then removed all changes from the modified `build_dataflow_steps.py` and retried building but to no avail. I have attached the DRC report.
As you were able to generate a bitstream for this model, how did you get around this particular resourcing issue? Thanks.
[top_wrapper_drc_opted.txt](https://github.com/Xilinx/finn/files/11557352/top_wrapper_drc_opted.txt)

KokyoK, 2023-05-25:
Hi @fionnodonohoe-xlnx ,
We tried again and did not meet your problem. I attached the `build_customize` folder, and we just tried this build with no error. 
This is the attachment: https://drive.google.com/file/d/1yfMkpSVOmBp5GrzpWn62daF1Um2d_-RO/view?usp=sharing