Xilinx/finn #801: "./run-docker.sh quicktest" on a fresh install fails some tests

sdittmeier, 2023-04-20:
I followed the instructions https://finn.readthedocs.io/en/latest/getting_started.html to set up FINN. 
I have installed VITIS 2022.1 and docker according to the documentation.

At step 4, some tests are failing.
Summary see below:

`FAILED tests/end2end/test_end2end_mobilenet_v1.py::test_end2end_mobilenet_export - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[Pynq-Z1-30] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_fpgadataflow_res_estimate.py::test_res_estimate - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_fpgadataflow_ipstitch.py::test_fpgadataflow_ipstitch_iodma_floorplan - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[Pynq-Z1-100000] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[Pynq-Z1-10000000] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[Ultra96-30] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[Ultra96-100000] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[Ultra96-10000000] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[U200-30] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[U200-100000] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/fpgadataflow/test_set_folding.py::test_set_folding[U200-10000000] - TypeError: __init__() got an unexpected keyword argument 'onnx_opset_version'
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[TFC-1-1] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/test_infer_datatypes_lfc.py::test_infer_datatypes_lfc - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/test_batchnorm_to_affine_bnn_pynq.py::test_batchnorm_to_affine_cnv_w1a1 - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/test_infer_data_layouts_cnv.py::test_infer_data_layouts_cnv - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/test_batchnorm_to_affine_bnn_pynq.py::test_batchnorm_to_affine_lfc_w1a1 - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[TFC-1-2] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_sign_to_thres.py::test_sign_to_thres - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_cnv.py::test_streamline_cnv[CNV-1-1] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[TFC-2-2] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_cnv.py::test_streamline_cnv[CNV-1-2] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[SFC-1-1] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[SFC-1-2] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_cnv.py::test_streamline_cnv[CNV-2-2] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[SFC-2-2] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[LFC-1-1] - TypeError: forward() takes 2 positional arguments but 5 were given
FAILED tests/transformation/streamline/test_streamline_fc.py::test_streamline_fc[LFC-1-2] - TypeError: forward() takes 2 positional arguments but 5 were given
ERROR tests/brevitas/test_brevitas_avg_pool_export.py
ERROR tests/brevitas/test_brevitas_cnv.py
ERROR tests/brevitas/test_brevitas_debug.py
ERROR tests/brevitas/test_brevitas_fc.py
ERROR tests/brevitas/test_brevitas_non_scaled_quanthardtanh_export.py
ERROR tests/brevitas/test_brevitas_qconv2d.py
ERROR tests/brevitas/test_brevitas_qlinear.py
ERROR tests/brevitas/test_brevitas_relu_act_export.py
ERROR tests/brevitas/test_brevitas_scaled_qhardtanh_export.py
ERROR tests/end2end/test_end2end_bnn_pynq.py
ERROR tests/end2end/test_end2end_cybsec_mlp.py
ERROR tests/fpgadataflow/test_fifosizing.py
ERROR tests/fpgadataflow/test_split_large_fifos.py
ERROR tests/transformation/test_qonnx_to_finn.py
====== 28 failed, 382 passed, 13 skipped, 1 xfailed, 4531 warnings, 14 errors in 20.04s ======`

Here is 1 more detailed error log

`Downloading: "https://github.com/Xilinx/brevitas/releases/download/bnn_pynq-r1/sfc_1w2a-52aacbed.pth" to /tmp/home_dir/.cache/torch/hub/checkpoints/sfc_1w2a-52aacbed.pth
100%|██████████| 1.30M/1.30M [00:00<00:00, 2.48MB/s]
________________________________ test_streamline_cnv[CNV-2-2] ________________________________
[gw5] linux -- Python 3.8.5 /opt/conda/bin/python

size = 'CNV', wbits = 2, abits = 2

    @pytest.mark.streamline
    # act bits
    @pytest.mark.parametrize("abits", [1, 2])
    # weight bits
    @pytest.mark.parametrize("wbits", [1, 2])
    # network topology / size
    @pytest.mark.parametrize("size", ["CNV"])
    def test_streamline_cnv(size, wbits, abits):
        if wbits > abits:
            pytest.skip("No wbits > abits cases at the moment")
        nname = "%s_%dW%dA" % (size, wbits, abits)
        finn_onnx = export_onnx_path + "/%s.onnx" % nname
        fc = get_test_model_trained(size, wbits, abits)
>       bo.export_finn_onnx(fc, (1, 3, 32, 32), finn_onnx)

tests/transformation/streamline/test_streamline_cnv.py:66: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
deps/brevitas/src/brevitas/export/__init__.py:18: in export_finn_onnx
    return FINNManager.export(*args, **kwargs)
deps/brevitas/src/brevitas/export/onnx/finn/manager.py:137: in export
    onnx_model = cls.export_onnx(
deps/brevitas/src/brevitas/export/onnx/manager.py:109: in export_onnx
    cls._cache_inp_out(module, *model_args, **model_kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

cls = <class 'brevitas.export.onnx.finn.manager.FINNManager'>
module = CNV(
  (conv_features): ModuleList(
    (0): QuantIdentity(
      (input_quant): ActQuantProxyFromInjector(
        (_...sentinel): StatelessBuffer()
      )
      (export_handler): FINNQuantLinearHandler()
    )
    (7): TensorNorm()
  )
)
args = (1, 3, 32, 32), kwargs = {}, stack = <contextlib.ExitStack object at 0x7fe3eddf5190>

    @classmethod
    def _cache_inp_out(cls, module, *args, **kwargs):
        # force enable caching
        module.apply(lambda m: _override_quant_metadata_caching_mode(m, enabled=True))
        module.apply(lambda m: _override_bias_caching_mode(m, enabled=True))
        module.apply(lambda m: _override_inp_caching_mode(m, enabled=True))
        module.apply(lambda m: _override_out_caching_mode(m, enabled=True))
        with ExitStack() as stack:
            for mgr in cls._cache_patches():
                stack.enter_context(mgr)
>           _ = module.forward(*args, **kwargs)
E           TypeError: forward() takes 2 positional arguments but 5 were given

deps/brevitas/src/brevitas/export/manager.py:263: TypeError
------------------------------------ Captured stderr call ------------------------------------`

auphelia, 2023-04-20:
Hi @sdittmeier,
Could you please provide more information:
- Are you working with the main branch? Or a specific commit version of FINN? 
- Did you change the `fetch_repos.sh` or are you working with the dependencies at the commit version in which they're specified?

For context:
The error is thrown in the Brevitas export function, so when the PyTorch model gets exported to FINN-ONNX format. It is important to use the Brevitas version that is linked in the `fetch_repos.sh`.

sdittmeier, 2023-04-20:
Hi @auphelia ,

I'm working with the main branch, output from `git log`

commit b3bdff118ae076cb776af6e51ddc28eeaa0d6390
Merge: cdc5ec4 9847528
Author: auphelia <56755897+auphelia@users.noreply.github.com>
Date:   Mon Feb 13 11:55:42 2023 +0000

    Merge pull request #762 from Xilinx/fix/nb_tests
    
    Fix known issues for release

I did not change anything, it was just a fresh clone of the repository. So `fetch_repos.sh` is in its state from the main branch.

Brevitas is cloned into the `deps` directory. 
What I can see is that the commit of the repo however does not match what is specified in the `fetch_repos.sh`
`BREVITAS_COMMIT="a5b71d6de1389d3e7db898fef72e014842670f03"` is specifed there
however, git log gives

commit c65f9c13dc124971f14739349531bbcda5c2a4aa
Author: Alessandro Pappalardo <alessand@amd.com>
Date:   Tue Jan 10 00:12:47 2023 +0000

    Docs: update README to v0.8.0
    
    Signed-off-by: Alessandro Pappalardo <alessand@amd.com>

It seems that this happened for the other dependencies as well. 
I'll try a manual checkout of the commits of all dependencies and report back if that fixed the issues.

auphelia, 2023-04-20:
Great, I am closing the issue then :) If you find out why different commits were used for the dependencies for your setup, please let us know!