Xilinx/finn #808: A problem with PYNQ board first-time setup

wanglu4042, 2023-04-28:
## Prerequisites
Hi, I am a rookie to FPGA , also with finn, sorry to bother you, but i did find some problems when I  was following the tutorial.

## Details
1. get into the finn folder, and do `bash ./run-docker.sh`, there is a line about the board which is 
[image]
Is setting the  `PLATFORM_REPO_PATHS` necessary? 
My board is PYNQ-Z2, how can I change `Using default PYNQ board Pynq-Z1` to Pynq-Z2?
2. When I did this, there was an error.
[image]
the error was 
[image]
How can I fix it?
I can skip it and do the next step `Test that you can ssh <PYNQ_USERNAME>@<PYNQ_IP> without having to enter the password. Pass the -v flag to the ssh command if it doesn’t work to help you debug.`, is it OK?

It's also my first time to submit an issue, if there is any problem, please tell me.

Looking forward to the answer!
Sincerely

closed 2023-05-07 by wanglu4042

Ba1tu3han, 2023-10-09:
@wanglu4042 Could you explain your solution please. Thank you! Bests!

wanglu4042, 2023-10-15:
> @wanglu4042 Could you explain your solution please. Thank you! Bests!

Hi, I didn't fix that issue, because it would not influence your project. I think  you can just ignore it.