Xilinx/finn #858: No module named 'finn.core' finn 0.0.1

MODISENEISO, 2023-07-18:
[image]
I am getting this error while going through tutorial /3-build-accelerator-with-finn.ipynb

auphelia, 2023-07-21:
Hi @MODISENEISO ,

Could you please use the issue template?
The error message looks to me like the Docker container was not set up correctly. Did you follow the steps on the [Getting Started](https://finn.readthedocs.io/en/latest/getting_started.html) page?

MODISENEISO, 2023-07-21:
> Hi @MODISENEISO ,
> 
> Could you please use the issue template? The error message looks to me like the Docker container was not set up correctly. Did you follow the steps on the [Getting Started](https://finn.readthedocs.io/en/latest/getting_started.html) page?

Yes I have started going this ruote however I have a challenge mounting the Vivado in the docker.The vivado is installed in a conda env. I added the path for installation before launching docker 
[image]

MODISENEISO, 2023-07-27:
> 
Good day I managed to run the image and have container instance. However the jupyter notebook after starting in the terminal, the link does not open in the browser. Any susgestions as to what could be the cause

MODISENEISO, 2023-07-27:
> > 
> 
> Good day I managed to run the image and have container instance. However the jupyter notebook after starting in the terminal, the link does not open in the browser. Any susgestions as to what could be the cause

MODISENEISO, 2023-07-27:
@auphelia

fpjentzsch, 2023-10-16:
Hi, the original issue seems to be resolved, so I'll close this one. If you still encounter issues with the Jupyter server, please check related discussions (e.g. https://github.com/Xilinx/finn/discussions/854) and create a new discussion if needed.

closed 2023-10-16 by fpjentzsch