Xilinx/finn #870: Finn fresh install with quicktest

smtalds, 2023-08-02:
Hi , i am trying to install finn on Ubuntu 18.04 and python 3.8.11. When i run quicktest , i got this error : 

[image]

I tried main version. Then i tried other version like v0.8.1 or v0.8 i got this error : I can not solve problems.

[image]

Thanks for answering.

fpjentzsch, 2023-08-02:
Hi, this dependency issue should be solved on the current dev branch. Can you try that one?

smtalds, 2023-08-02:
Hi again @fpjentzsch , yes its worked. I can open notebook , but i have still 4xfailed . Is it important ? 

[image]

Thanks for answering , maybe u can pull this branch for fresh installing. If this is not important , we can close issue. Thanks for everything.

fpjentzsch, 2023-08-02:
No, these tests are currently marked as "expected fail", so there is no problem.

closed 2023-08-02 by fpjentzsch

smtalds, 2023-08-07:
Hi again @fpjentzsch , I worked on dev branch now , (this is my second pc). I can run same ipynb file my second pc (i installed finn on feb or before then so i dont know which branch), but i can not run same ipynb my first pc (dev branch.) This is error :

[image]
 Can u help me again , thank you sir.

smtalds, 2023-08-07:
Actually i dont know why x is a list. Then , i tried and export_qonnx is going to same function and i got same error.

smtalds, 2023-08-18:
I found the answer on https://github.com/Xilinx/finn/issues/853#issuecomment-1640105304. You have to use ishape with torch.rand(ishape).

Ba1tu3han, 2023-09-16:
> Hi, this dependency issue should be solved on the current dev branch. Can you try that one?

Hello could you explain to me how to use the dev brach? I directly pull the repo that is written in the getting started page. Thanks

smtalds, 2023-09-17:
> > Hi, this dependency issue should be solved on the current dev branch. Can you try that one?
> 
> Hello could you explain to me how to use the dev brach? I directly pull the repo that is written in the getting started page. Thanks

You can use this command 
`git clone -b <branchname> <remote-repo-url>`  , branchname should be "dev"

Ba1tu3han, 2023-09-18:
> You can use this command `git clone -b <branchname> <remote-repo-url>` , branchname should be "dev"

It works for me, too. Thank you!