Xilinx/finn #876: Throughput test result for clok

smtalds, 2023-08-23:
Hi , i build the confing like this settings : 
[image]
and my troughput results is showing clock 100MHz.
[image]
But I opened the vivado project , clock is 200MHz. So its true. Is it a bug or problem ?
[image] Btw , I have so much TNS. What is the problem ? How can i solve this problem ?  Thanks for answer.
[image]

fpjentzsch, 2023-08-29:
Hi,

the design was indeed synthesized with a 200 MHz clock because you set `synth_clk_period_ns=5.0` in the build config.

The throughput test shows 100 MHz, because you did not specify a clock with the `fclk_mhz` argument when initializing the `FINNExampleOverlay`, so it defaults to 100 MHz. The PYNQ driver can reconfigure the PL clock generator at runtime by changing the divisors, and this clock configuration is performed every time you load the overlay/bitstream by instantiating the `FINNExampleOverlay`.

I hope this helps, maybe you can try to synthesize for 100 MHz to improve your timing.

smtalds, 2023-08-29:
@fpjentzsch  Ah , I was thinking `FINNExampleOverlay` is get `fclk` from bitfile. It was my fault :/ Thanks for answering.

closed 2023-08-29 by smtalds