Xilinx/finn #887: run-docker.sh has Pip dependency conflict on wsl2 ubuntu 22.04

canxin121, 2023-09-08:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [√] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [√ ] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
I just run the ./run-docker.sh notebook on my wsl2 ubuntu 22.04,and this error occurs. 
```python
Found Vitis HLS at /mnt/d/xilinx//Vitis_HLS/2023.1
Traceback (most recent call last):
  File "/opt/conda/bin/jupyter-notebook", line 5, in <module>
    from notebook.app import main
  File "/opt/conda/lib/python3.8/site-packages/notebook/app.py", line 9, in <module>
    from jupyter_server.base.handlers import JupyterHandler
  File "/opt/conda/lib/python3.8/site-packages/jupyter_server/base/handlers.py", line 23, in <module>
    from jupyter_events import EventLogger
  File "/opt/conda/lib/python3.8/site-packages/jupyter_events/__init__.py", line 3, in <module>
    from .logger import EVENTS_METADATA_VERSION, EventLogger
  File "/opt/conda/lib/python3.8/site-packages/jupyter_events/logger.py", line 13, in <module>
    from jsonschema import ValidationError
  File "/opt/conda/lib/python3.8/site-packages/jsonschema/__init__.py", line 13, in <module>
    from jsonschema._format import FormatChecker
  File "/opt/conda/lib/python3.8/site-packages/jsonschema/_format.py", line 11, in <module>
    from jsonschema.exceptions import FormatError
  File "/opt/conda/lib/python3.8/site-packages/jsonschema/exceptions.py", line 14, in <module>
    from attrs import define
  File "/opt/conda/lib/python3.8/site-packages/attrs/__init__.py", line 3, in <module>
    from attr import (
ImportError: cannot import name 'AttrsInstance' from 'attr' (/tmp/home_dir/.local/lib/python3.8/site-packages/attr/__init__.py)
```

Ba1tu3han, 2023-09-16:
I have the same attr library issue. I could not solve it. https://github.com/Xilinx/finn/discussions/889

smtalds, 2023-09-27:
@gb16001 and @canxin121 Check #870 , we used dev branch for this issue and it solved.

gb16001, 2023-09-30:
it works for me, thanks a lot

closed 2023-10-16 by fpjentzsch