Xilinx/finn #891: a bug about attrs

wz19980621, 2023-09-16:
ERROR: After October 2020 you may experience errors when installing or updating packages. This is because pip will change the way that it resolves dependency conflicts.

We recommend you use --use-feature=2020-resolver to test your packages with the new resolver before it becomes the default.

referencing 0.30.2 requires attrs>=22.2.0, but you'll have attrs 19.3.0 which is incompatible.
jsonschema 4.19.0 requires attrs>=22.2.0, but you'll have attrs 19.3.0 which is incompatible.

Ba1tu3han, 2023-09-16:
I have the same issue. Please let me know when you figure it out. https://github.com/Xilinx/finn/discussions/889

wz19980621, 2023-09-22:
> I have the same issue. Please let me know when you figure it out. #889
> 
> @wz19980621 did you check this issue? #870
> 
> I found a website that mentiones about this attrs dependency issue. https://www.daiphys.com/portal/fpga/xilinx/tools/finn.html

Thanks for your answer, it's very useful

closed 2023-09-22 by wz19980621