Xilinx/finn #906: Empty Jupyter Notebook Page

Ba1tu3han, 2023-10-11:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [+] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [+] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
When I run "bash ./run-docker.sh notebook " command. I connect to http://127.0.0.1:8888/tree but this page is empty although I was running and accessing notebook as expected this morning. 

## Details

I did not change any thing. I just pulled the repo and run it well. I wanted to run it a notebook again and I encountered fully empty page. I restarted everything. I still can not access notebooks in Jupyter notebook. 

[image]

1. Clone the FINN repository
2. Checkout the dev branch, with commit hash: [13313c2]
3. Start the docker container with the command: [bash ./run-docker.sh notebook]
4. Run transformation [...] on ONNX file [...] or run the dataflow builder with the following settings: [...]
5. [Further steps ...]

### Expected behavior
Accessing notebooks

### Actual behavior
Empty Jupyter Notebook page

## Optional

### Possible fix
If you already know where the issue stems from, or you have a hint please let us know.

### Additional context
Add any other context about the problem here.

fpjentzsch, 2023-10-16:
Hi, does the Jupyter server console output show any error message when you try to connect via the browser?

Ba1tu3han, 2023-10-17:
Hi, unfortunatly no, I have not any error message. 
[image]