Xilinx/finn #933: Unable to launch FINN docker container

Madhurima8, 2023-12-05:
### Discussed in https://github.com/Xilinx/finn/discussions/932

<div type='discussions-op-text'>

<sup>Originally posted by **Madhurima8** December  3, 2023</sup>
I am facing an issue while installing FINN with docker. when I ran this command to verify the installation – ./run-docker.sh quicktest, I got the below error -
[image]

Is there a way to resolve this?

</div>

deremustapha, 2023-12-28:
Try following this instructions on here: https://www.daiphys.com/portal/fpga/xilinx/tools/finn.html. I think this would resolve the problem.

auphelia, 2024-04-03:
Thank you @deremustapha for the suggestion!
Before closing the issue, I would like to recommend using the latest tag and trying it again. There were some updates to the setup.py file with the latest release.