Xilinx/finn #948: ERROR: [HLS 200-1023] Part 'xczu7ev-2ffvc1156' is not installed.

ppccww0201, 2024-01-05:
Hi,
I am following the [third example](https://github.com/Xilinx/finn/blob/main/notebooks/end2end_example/cybersecurity/3-build-accelerator-with-finn.ipynb). I am going through the examples one by one, and I encountered an error while building the accelerator.

[image]

Here is error message
```
Building dataflow accelerator from /home/vboxuser/Desktop/FINN/finn-examples/build/finn/notebooks/end2end_example/cybersecurity/cybsec-mlp-ready.onnx
Intermediate outputs will be generated in /tmp/finn_dev_vboxuser
Final outputs will be generated in output_ipstitch_ooc_rtlsim
Build log is at output_ipstitch_ooc_rtlsim/build_dataflow.log
Running step: step_qonnx_to_finn [1/18]
Running step: step_tidy_up [2/18]
Running step: step_streamline [3/18]
Running step: step_convert_to_hls [4/18]
Running step: step_create_dataflow_partition [5/18]
Running step: step_target_fps_parallelization [6/18]
Running step: step_apply_folding_config [7/18]
Running step: step_minimize_bit_width [8/18]
Running step: step_generate_estimate_reports [9/18]
Running step: step_hls_codegen [10/18]
Running step: step_hls_ipgen [11/18]

multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "/home/vboxuser/Desktop/FINN/finn-examples/build/finn/src/finn/transformation/fpgadataflow/hlssynth_ip.py", line 71, in applyNodeLocal
    inst.ipgen_singlenode_code()
  File "/home/vboxuser/Desktop/FINN/finn-examples/build/finn/src/finn/custom_op/fpgadataflow/hlscustomop.py", line 356, in ipgen_singlenode_code
    assert os.path.isdir(ip_path), "IPGen failed: %s not found. Check log under %s" % (
AssertionError: IPGen failed: /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/project_MatrixVectorActivation_2/sol1/impl/ip not found. Check log under /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/vboxuser/Desktop/FINN/finn-examples/build/finn/src/finn/builder/build_dataflow.py", line 161, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/home/vboxuser/Desktop/FINN/finn-examples/build/finn/src/finn/builder/build_dataflow_steps.py", line 499, in step_hls_ipgen
    model = model.transform(HLSSynthIP())
  File "/home/vboxuser/Desktop/FINN/finn-examples/build/finn/deps/qonnx/src/qonnx/core/modelwrapper.py", line 140, in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
  File "/home/vboxuser/Desktop/FINN/finn-examples/build/finn/deps/qonnx/src/qonnx/transformation/base.py", line 111, in apply
    new_nodes_and_bool = p.map(self.applyNodeLocal, old_nodes, chunksize=1)
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 774, in get
    raise self._value
AssertionError: IPGen failed: /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/project_MatrixVectorActivation_2/sol1/impl/ip not found. Check log under /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8
```

And here is log:
```

****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2022.1 (64-bit)
  **** SW Build 3526262 on Mon Apr 18 15:47:01 MDT 2022
  **** IP Build 3524634 on Mon Apr 18 20:55:01 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vitis_HLS/2022.1/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vitis_HLS/2022.1/bin/unwrapped/lnx64.o/vitis_hls'
/tools/Xilinx/Vitis_HLS/2022.1/tps/tcl/tcl8.5/tzdata/Europe/Dublin can't be opened.
INFO: [HLS 200-10] For user 'vboxuser' on host 'finn_dev_vboxuser' (Linux_x86_64 version 5.4.0-150-generic) on Fri Jan 05 11:44:57 +0000 2024
INFO: [HLS 200-10] On os Ubuntu 22.04.1 LTS
INFO: [HLS 200-10] In directory '/tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8'
Sourcing Tcl script '/tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/hls_syn_MatrixVectorActivation_2.tcl'
INFO: [HLS 200-1510] Running: source /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/hls_syn_MatrixVectorActivation_2.tcl
HLS project: project_MatrixVectorActivation_2
HW source dir: /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8
finn-hlslib dir: /home/vboxuser/Desktop/FINN/finn-examples/build/finn/deps/finn-hlslib
custom HLS dir: /home/vboxuser/Desktop/FINN/finn-examples/build/finn/custom_hls
INFO: [HLS 200-1510] Running: open_project project_MatrixVectorActivation_2 
INFO: [HLS 200-10] Creating and opening project '/tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/project_MatrixVectorActivation_2'.
INFO: [HLS 200-1510] Running: add_files /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/top_MatrixVectorActivation_2.cpp -cflags -std=c++14 -I/home/vboxuser/Desktop/FINN/finn-examples/build/finn/deps/finn-hlslib -I/home/vboxuser/Desktop/FINN/finn-examples/build/finn/custom_hls 
INFO: [HLS 200-10] Adding design file '/tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/top_MatrixVectorActivation_2.cpp' to the project
INFO: [HLS 200-1510] Running: set_top MatrixVectorActivation_2 
INFO: [HLS 200-1510] Running: open_solution sol1 
INFO: [HLS 200-10] Creating and opening solution '/tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/project_MatrixVectorActivation_2/sol1'.
INFO: [HLS 200-1505] Using default flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2022.1;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-435] Setting 'open_solution -flow_target vivado' configuration: config_interface -m_axi_latency=0
INFO: [HLS 200-1510] Running: set_part xczu7ev-2ffvc1156 
ERROR: [HLS 200-1023] Part 'xczu7ev-2ffvc1156' is not installed.
command 'create_platform' returned error code
    while executing
"source /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/hls_syn_MatrixVectorActivation_2.tcl"
    invoked from within
"hls::main /tmp/finn_dev_vboxuser/code_gen_ipgen_MatrixVectorActivation_2_4npd16v8/hls_syn_MatrixVectorActivation_2.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$newargs"
    (procedure "hls_proc" line 16)
    invoked from within
"hls_proc [info nameofexecutable] $argv"
INFO: [HLS 200-112] Total CPU user time: 0.78 seconds. Total CPU system time: 0.44 seconds. Total elapsed time: 0.8 seconds; peak allocated memory: 612.438 MB.
INFO: [Common 17-206] Exiting vitis_hls at Fri Jan  5 11:44:57 2024...
```

I'm targeting ZCU-104, so I changed code like below..
```
cfg_stitched_ip = build.DataflowBuildConfig(
    output_dir          = rtlsim_output_dir,
    mvau_wwidth_max     = 80,
    target_fps          = 1000000,
    synth_clk_period_ns = 10.0,
    fpga_part           = "xczu7ev-2ffvc1156", # ZCU-104 fpga part
    generate_outputs=[
        build_cfg.DataflowOutputType.STITCHED_IP,
        build_cfg.DataflowOutputType.RTLSIM_PERFORMANCE,
        build_cfg.DataflowOutputType.OOC_SYNTH,
    ]
)
```

Here's my environment info:
`Ubuntu 18.04`
Vitis, Vitis_HLS, Vivado version is `2022.1`
finn repo branch is `dev`

I don't know how to handle error `ERROR: [HLS 200-1023] Part 'xczu7ev-2ffvc1156' is not installed.`

Please let me know how to resolve this error!! Thank you!

bwintermann, 2024-01-05:
This means the board files for that part are missing. These files can be found here for example: https://github.com/Xilinx/XilinxBoardStore/tree/2022.2/boards/Xilinx/zcu104/1.1

Instructions on how to install these files can be found here https://github.com/Xilinx/XilinxBoardStore/wiki/Accessing-the-Board-Store-Repository though a manual installation should also be possible (as in https://support.xilinx.com/s/question/0D54U00007TQ6f8SAD/how-to-fix-error-hls-2001023-part-is-not-installed-?language=en_US)

ppccww0201, 2024-01-08:
Thank you for your reply.

I think I already have board files..
[image]
(path is `/tools/Xilinx/Vivado/2022.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/zcu104/1.1`)

Is this not enought to prevent this error `ERROR: [HLS 200-1023] Part 'xczu7ev-2ffvc1156' is not installed.`?

Or do I need something more process like "registration" thing?

ppccww0201, 2024-01-08:
> This means the board files for that part are missing. These files can be found here for example: https://github.com/Xilinx/XilinxBoardStore/tree/2022.2/boards/Xilinx/zcu104/1.1
>
> Instructions on how [...]

Thank you for your reply.

I think I already have board files..
[image]
(path is `/tools/Xilinx/Vivado/2022.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/zcu104/1.1`)

Is this not enought to prevent this error `ERROR: [HLS 200-1023] Part 'xczu7ev-2ffvc1156' is not installed.`?

Or do I need something more process like "registration" thing?

bwintermann, 2024-01-08:
Could you try to change your part name to "xczu7ev-ffvc1156-2-e"? This is also the part name that is used in the board.xml of the zcu104/1.1 board files.

closed 2024-01-08 by ppccww0201