Xilinx/finn #954: ConCat layer Issue

pkdeep, 2024-01-15:
Hi @williamliao28 ,
I am facing this issue where Concat layer is not getting converted to HLS layer. I tried the transformation InferConCatLayer(), but still it dint infer streaming Concat layer. Can you help with this. I am attacing original and converted .onnx files
[image]

Thanks for raising this issue! I will link this conversation in the GitHub discussion you started initially.

We merged a patch in qonnx and updated the latest qonnx version in `dev`, please try it out and let me know if there are any problems. Closing this issue for now, but please feel free to re-open it if necessary.

One remark because I had a look at your generated files, you should call a few additional streamlining functions to get rid of the transpose nodes (e.g. [AbsorbConsecutiveTransposes](https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/streamline/absorb.py#L453)) and also to make the necessary data layout conversions, like [MakeMaxPoolNHWC](https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/streamline/reorder.py#L626).

_Originally posted by @auphelia in https://github.com/Xilinx/finn/issues/853#issuecomment-1640403156_

iksnagreb, 2024-01-22:
There are scalar multiplications of floating-point type preceding the inputs to the Concat operator, turning the inputs into floating-point as well. Currently, the StreamingConcat does not support floating-point inputs, see here: https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/fpgadataflow/convert_to_hls_layers.py#L1659

Unfortunately, this seems not to be easy to streamline, as the scale factors on each input branch are different. Streamlining the Mul nodes past the Concat layer only works out if they are all equal... It could work using some group-wise scale-factors afterwards, but I am not sure whether there is support for this.

Allowing the StreamingConcat operation to handle floating-point inputs is also not really desirable, as it would leave floating-points in the middle of your model graph (with finn you ideally want to have a continuous chain of integer-only operations). While it should be technically possible, this would require some refactoring of the corresponding HLS code generation.

pkdeep, 2024-01-23:
Hi @iksnagreb . Thanks for the reply. I could get the Mul factors to be equal. Can this be processed to have a streaming replacement?
 
 I am attaching both the pre-dataflow creation (after HLS) model image and after dataflow creation image. Any insight would be of great help to convert this into a fully streamlined model.

[image]
[image]

iksnagreb, 2024-01-23:
Hm, conceptually, you now would need something like the `MoveIdenticalOpPastJoinOp` transformation (this is in `finn.transformation.streamline.reorder`), but it seems to be made for two-input join-nodes, while you have an "arbitrary" many inputs join-node. I am not sure whether there already is a suitable streamlining transformation somewhere in finn, but adapting the existing `MoveIdenticalOpPastJoinOp` or introducing a new, similar one seems not too complicated.

auphelia, 2024-04-03:
Closing this issue, thanks @iksnagreb for the suggestions. @pkdeep, if you have tried out to extend `MoveIdenticalOpPastJoinOp` and would like to contribute to the finn compiler, we're always welcoming contributions from the community 🙂
Please refer to the [contribution guidelines](https://github.com/Xilinx/finn/blob/main/CONTRIBUTING.md) in that case.

closed 2024-04-03 by auphelia

pkdeep, 2024-09-28:
Hi,
Sorry for the delayed response. I was able to move the uniform Multiplication past the Concat layer, although not by "MoveIdenticalOpPastJoinOp" but through some other technique. I removed the mul node, and inserted a new mul node after Concat and the put the exact same value of Param as the deleted Mul node. This helped me in able to convert the model in end-to-en streaming model (except the last Mul and Add node). Thanks
To get uniform Mul node, I inserted "Quantidentity" in each of 4 branches. It gave me same multiplication factor in all 4 branches, but I am not clear why it happened? Can you help in understanding the reason?