Xilinx/finn #955: cycle-free graph violated

ppccww0201, 2024-01-16:
Hi!

I'm trying to build my custom model accelerator based on [end2end_example/cybersecurity](https://github.com/Xilinx/finn/tree/dev/notebooks/end2end_example/cybersecurity).

My custom model is below:
```
class QuantMobileNet(nn.Module):
    def __init__(self, nclasses):
        super().__init__()
        self.nclasses = nclasses
        self.Act_Quant_Bit = 4
        self.return_quant_tensor = True

        self.identity_inp = qnn.QuantIdentity(act_quant=ActQuant, bit_width=self.Act_Quant_Bit, return_quant_tensor=self.return_quant_tensor)

        self.conv0 = qnn.QuantConv2d(1, 64, kernel_size=(1,1), bias=False)
        self.bn0 = nn.BatchNorm2d(64)
        self.relu0 = nn.ReLU()
        
        self.conv1 = qnn.QuantConv2d(64, 64, kernel_size=5, stride=2, groups=64, bias=False)
        self.bn1 = nn.BatchNorm2d(64)
        self.relu1 = qnn.QuantReLU(bit_width=self.Act_Quant_Bit)
        self.conv2 = qnn.QuantConv2d(64, 128, kernel_size=1, stride=1, padding=0, bias=False)
        self.bn2 = nn.BatchNorm2d(128)
        self.relu2 = qnn.QuantReLU(bit_width=self.Act_Quant_Bit)
        self.conv3 = qnn.QuantConv2d(128, 128, kernel_size=5, stride=2, groups=128, bias=False)
        self.bn3 = nn.BatchNorm2d(128)
        self.relu3 = qnn.QuantReLU(bit_width=self.Act_Quant_Bit)
        self.conv4 = qnn.QuantConv2d(128, 256, kernel_size=1, stride=1, padding=0, bias=False)
        self.bn4 = nn.BatchNorm2d(256)
        self.maxpool = nn.MaxPool2d(kernel_size=4)
        self.fc = qnn.QuantLinear(256, nclasses, bias=True)

    def forward(self, x):
        xi = self.identity_inp(x)
        xc0 = self.conv0(xi)
        xb0 = self.bn0(xc0)
        xr0 = self.relu0(xb0)  
        xc1 = self.conv1(xr0)
        xb1 = self.bn1(xc1)
        xr1 = self.relu1(xb1)
        xc2 = self.conv2(xr1)
        xb2 = self.bn2(xc2)
        xr2 = self.relu2(xb2)
        xc3 = self.conv3(xr2)
        xb3 = self.bn3(xc3)
        xr3 = self.relu3(xb3)
        xc4 = self.conv4(xr3)
        xb4 = self.bn4(xc4)
        xp = self.maxpool(xb4)
        x_flat = xp.view(xp.size(0), -1)
        xc5 = self.fc(x_flat)
        
        return xc5
```

and each layer shape is below:
```
xc0: torch.Size([1, 64, 28, 28])
xb0: torch.Size([1, 64, 28, 28])
xr0: torch.Size([1, 64, 28, 28])
xc1: torch.Size([1, 64, 12, 12])
xb1: torch.Size([1, 64, 12, 12])
xr1: torch.Size([1, 64, 12, 12])
xc2: torch.Size([1, 128, 12, 12])
xb2: torch.Size([1, 128, 12, 12])
xr2: torch.Size([1, 128, 12, 12])
xc3: torch.Size([1, 128, 4, 4])
xb3: torch.Size([1, 128, 4, 4])
xr3: torch.Size([1, 128, 4, 4])
xc4: torch.Size([1, 256, 4, 4])
xb4: torch.Size([1, 256, 4, 4])
xp: torch.Size([1, 256, 1, 1])
x_flat: torch.Size([1, 256])
xc5: torch.Size([1, 10])
```

and then I export my model as onnx format with below code:
```
model_for_export = Quant_MobileNet # Quant_MobileNet is trained model

from brevitas.export import export_qonnx
from qonnx.util.cleanup import cleanup as qonnx_cleanup
from qonnx.core.modelwrapper import ModelWrapper
from qonnx.core.datatype import DataType
from finn.transformation.qonnx.convert_qonnx_to_finn import ConvertQONNXtoFINN
from finn.transformation.streamline.reorder import MakeMaxPoolNHWC

ready_model_filename = model_dir + "/LeNet.onnx"
input_shape = (1, 1, 28, 28)

#Move to CPU before export
model_for_export.cpu()

# Export to ONNX
export_qonnx(
    model_for_export, export_path=ready_model_filename, input_t=torch.randn(input_shape)
)

# clean-up
qonnx_cleanup(ready_model_filename, out_file=ready_model_filename)

# ModelWrapper
model = ModelWrapper(ready_model_filename)
# model = model.transform(MakeMaxPoolNHWC())
# Setting the input datatype explicitly because it doesn't get derived from the export function
print(images.dtype)
model.set_tensor_datatype(model.graph.input[0].name, DataType["FLOAT32"])
model = model.transform(ConvertQONNXtoFINN())
model.save(ready_model_filename)

print("Model saved to %s" % ready_model_filename)
```

Here is graph:
[image]

Finally, I tried to build like below:
```
import finn.builder.build_dataflow as build
import finn.builder.build_dataflow_config as build_cfg
import os
import shutil

model_file = model_dir + "/LeNet.onnx"

rtlsim_output_dir = "output_ipstitch_ooc_rtlsim"

#Delete previous run results if exist
if os.path.exists(rtlsim_output_dir):
    shutil.rmtree(rtlsim_output_dir)
    print("Previous run results deleted!")

cfg_stitched_ip = build.DataflowBuildConfig(
    verify_save_rtlsim_waveforms = True,
    output_dir          = rtlsim_output_dir,
    mvau_wwidth_max     = 80,
    target_fps          = 1000,
    synth_clk_period_ns = 10.0,
    fpga_part           = "xczu7ev-ffvc1156-2-e", # xczu7ev-2ffvc1156 xc7z020clg400-1
    generate_outputs=[
        build_cfg.DataflowOutputType.STITCHED_IP,
        build_cfg.DataflowOutputType.RTLSIM_PERFORMANCE,
        build_cfg.DataflowOutputType.OOC_SYNTH,
    ]
)

%%time
build.build_dataflow_cfg(model_file, cfg_stitched_ip)
```

but `AssertionError: cycle-free graph violated: partition depends on itself` error is occurred!
[image]

I already check the other's same problem, but it's not helpful due to my lack of understanding about finn.
Can you tell me how to solve my problem?

Thank you!

ppccww0201, 2024-01-17:
`nn.MaxPool2d(kernel_size=4)` is not certainly supported!!

And I think `groups` option of `qnn.QuantConv2d` is not supported too. (Not certain...)

gabrielpgagne, 2024-01-19:
If I recall, I had a similar error message when trying to build my model and it was because it was too big for my FPGA part (xc7z020). I reduced its size and the error went away. Maybe I'm mixing things up.

bwintermann, 2024-01-19:
> `nn.MaxPool2d(kernel_size=4)` is not certainly supported!!
> 
> And I think `groups` option of `qnn.QuantConv2d` is not supported too. (Not certain...)

Not sure if you've already found it, but this was talked about in an older thread: https://github.com/Xilinx/finn/discussions/431
It also contains a possible solution for you to try (in the form of custom steps with additional transformations) , maybe this helps.

closed 2024-01-25 by ppccww0201