Xilinx/finn #1031: Compiler Difference Between W1A1 and W4A4

Ba1tu3han, 2024-04-06:
## Quick summary
FINN v0.9 and v0.10 work with W1A1 but for W4A4 I encounter with an error. Which line should I change depending on binary or non-binary quantization? 

## Details
I use ONNX file as input that is generated by Brevitas.

### Steps to Reproduce
copy CNV end-to-end example and change the ONNX file
use W1A1 and W4A4 QAT by Brevitas

FINN: v0.10
Brevitas: 0.10.2
Ubuntu: 20.04
Docker: 25.0.1
Vivado: 2022.1

### Expected behavior
compile all ONNX files.

### Actual behavior
Does not compile for non-binary quantizations

[image]

auphelia, 2024-04-09:
Hi @Ba1tu3han,

The error is thrown when trying to convert a MatMul + MultiThreshold to an MVAU. From your description I assume that the MultiThreshold's purpose is a re-quantization to signed int 4. A signed activation should always have a negative bias, which is called `actval` in the code above but is derived here in the code: https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/fpgadataflow/convert_to_hw_layers.py#L1474
For the bipolar case this is different, because even though it counts as "signed datatype" the conversion from binary->bipolar is achieved by reinterpretation and so the actval (or bias) doesn't need to be negative.

So, there is a mismatch from what FINN expects to get as an input from the Brevitas model and how the actual model looks like. I see that you are using the latest Brevitas, we have not moved to this release yet in FINN. You might want to try using the same Brevitas version we currently use, you can find the commit hash here: https://github.com/Xilinx/finn/blob/dev/fetch-repos.sh#L32

Ba1tu3han, 2024-04-09:
Hello Auphelia,

Thank you for your reply. I will try it with the FINN version and check the transformation.

Best Regards,

closed 2024-08-07 by Ba1tu3han

Ba1tu3han, 2024-08-28:
Hello @auphelia ,

What is the situation right now for FINN and Brevitas versions? Can I use the latest ones for both?

Bests,

auphelia, 2024-08-28:
Hi @Ba1tu3han,
With FINN we are using a specific commit version of Brevitas (https://github.com/Xilinx/finn/blob/dev/fetch-repos.sh#L32) which is **currently** Brevitas version 0.10.0
We are working on updating the Brevitas commit hash, see here: https://github.com/Xilinx/finn/pull/1171