Xilinx/finn #1033: Strided pointwise convolution on rectangular feature maps not correctly handled by InferConvInpGen

hleblevec, 2024-04-08:
## Quick summary
The tool does not correctly handle strided pointwise convolution with rectangular feature maps anymore. With the previous build, I would use the RTL variant of ConvInpGen which supports this config, but now the transformation is trying to infer a DownSampler node because it detects a strided pointwise Im2Col, but the DownSampler node does not work with rectangular input.

## Details

### Steps to Reproduce
1. Use main branch with release 0.10.0
2. Have a model with a Convolution having stride>1 and kernel_size=1 but with non-square feature maps.
3. Decompose into Im2Col + MatMul in the streamlining step.
4. Run `step_convert_to_hw_layers`.

### Expected behavior
ConvInpGen node created with this configuration.

### Actual behavior
Triggers a warning: https://github.com/Xilinx/finn/blob/e3087ad9fbabcc35f21164d415ababec4f462e9f/src/finn/transformation/fpgadataflow/convert_to_hw_layers.py#L133

## Optional

### Possible fix
1. Is it still relevant to have a DownSampler node for the specific case of strided pointwise convolutions? Does it work better than a ConvInpGen in this specific scenario?
2. If yes, a possible fix would be, in the case of a rectangular input, to move to normal ConvInpGen in the `InferConvInpGen` transformation rather than skipping the conversion. This node would need to be rewired to the RTL variant in the `SpecializeLayers` transformation, which shouldn't be a problem because the `_swg_hls_possible` function verifies that the input is square.
https://github.com/Xilinx/finn/blob/e3087ad9fbabcc35f21164d415ababec4f462e9f/src/finn/transformation/fpgadataflow/specialize_layers.py#L201-L233

auphelia, 2024-04-10:
Hi @hleblevec,
Thanks for pointing that out! I actually think we could integrate the functionality of the DownSampler into the ConvolutionInputGenerator custom op structure as one additional variant. This would make this extra node obsolete. Let me know if that would be something you would like to contribute to the repo yourself!

hleblevec, 2024-04-12:
Hi @auphelia,
Yeah sure! If I understood well, it would be a variant introduced by `SpecializeLayers` depending on the given parameters?