Xilinx/finn #1034: No such file or directory: '/tmp/.../synth_out_of_context_d9pkqpve/results_finn_design_wrapper/res.txt'

Ba1tu3han, 2024-04-09:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [ +] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [+ ] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [ +] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
I run[ 3-build-accelerator-with-finn.ipynb](https://github.com/Xilinx/finn/blob/e3087ad9fbabcc35f21164d415ababec4f462e9f/notebooks/end2end_example/cybersecurity/3-build-accelerator-with-finn.ipynb) with a custom W1A1 ONNX file that is exported by Brevitas 0.10.2. I encountered an error which is shown below. I did not change anything else.

[image]
[QONNX_CNV.zip](https://github.com/Xilinx/finn/files/14923395/QONNX_CNV.zip)

### Steps to Reproduce
Add what needs to be done to reproduce the bug. Add code examples where useful
and make sure to include the resulting ONNX files, and the commit hash you are working on.

1. Clone the FINN v0.10 repository
2. Checkout the dev branch, with commit hash: [e3087ad9fbabcc35f21164d415ababec4f462e9f]
3. Start the docker container with the command: [bash ./run-docker.sh notebook]
4. Run transformation [...] on ONNX file [...] or run the dataflow builder with the following settings: [...]
5. Run the notebook by just changing the ONNX file.

### Expected behavior
Completing the procedure

Best Regards,

auphelia, 2024-04-12:
Hi @Ba1tu3han, this looks like synthesis failed. You can check out the log files in the directory that is linked in the error message.
The finn compiler cannot propagate automatically the error messages thrown by Vivado/Vitis. That is why you usually have errors like `No such file or directory` because after for example synthesis was executed, to verify that the run completed, the finn compiler checks if the expected files were generated and throws an error if not.

pbk20191, 2024-04-13:
This could happen if the design request more resource than target platform capacity( too much LUT, FliopFlop ... etc ).
Also check estmation report's resource part to ensure everything is under capacity

Ba1tu3han, 2024-04-18:
> This could happen if the design request more resource than target platform capacity( too much LUT, FlipFlop ... etc ). Also check estimation report's resource part to ensure everything is under capacity

Hello @pbk20191 and @auphelia 

Thank you for your comments. I check the log file and you are right. I exceed the LUT resource almost its double. 

I used the given CNV model in the Brevitas repo. I did not change any parameter. And In the finn-example repo Pynq-Z1 supports the CNV model. I use Pynq-Z2 board however, Z1 and Z2 have the exactly same FPGA. 

Did you @auphelia use a specific folding configurations for Pynq board for CNV model? I use auto folding (the basic one).

Do you have any recommendation? The LUT resource of Pynq-Z2 is 53.200 and estimated utilization is 111k and the log says it is around 127k ?

[image]
[image]

Ba1tu3han, 2024-07-03:
> Hi @Ba1tu3han, this looks like synthesis failed. You can check out the log files in the directory that is linked in the error message. The finn compiler cannot propagate automatically the error [...]

Hello, I could not solve it. This is the vivado log.

Could you help me?

[image]
[image]
[vivado.zip](https://github.com/user-attachments/files/16089724/vivado.zip)

KanishR1, 2024-10-04:
@auphelia  Hi, I am also facing the same issue. I opened the folder where vivado project files are there. Then I opened vivado and opened that project and implemented it. Implementation is successful and my resource utilization is less than the maximum available in ZCU102. Still why am I facing the issue when I do the same thing with finn's step_out_of_context_synthesis. Could you help me out with this error ?