Xilinx/finn #1039: Pass arguments to build_custom flow script

bwintermann, 2024-04-12:
## Details
### New behavior
Change the `run-docker.sh` script to accept any number of arguments for the build_custom option, so that these arguments can be passed along to the build script.

### Motivation
If the build python script is supposed to take arguments (for example to switch which steps to execute, target fps, etc.) and the interactive docker session is not an option, the only possibility right now to do this would be to either create different build scripts depending on which behaviour is wanted (but this gets unwieldy quickly) or to for example pass those args as an environment variable, but this more of a workaround, and since the arguments after `$3` are not used right now, it would be nice to pass them directly.

### Parts of FINN affected
`run-docker.sh`

alexhornburg-xlnx, 2024-11-19:
This was added in https://github.com/Xilinx/finn/pull/1217

auphelia, 2024-11-19:
Thanks @alexhornburg-xlnx for adding this feature! I am closing the issue for now, but @bwintermann feel free to reopen or create a new issue if this doesn't solve it for you.

closed 2024-11-19 by auphelia