Xilinx/finn #1055: The model works with V9 but not with new RTL layers in V10.

Arbiter-glitch, 2024-04-23:
[image]
[image]
There is a considerable difference between the output and the expected output. Expected output is the output image received while passing the input image through the software model. I use latest FINN branch. I trained the network outside the Finn docker. Then exported the qonnx format to Finn. Most of the nodes are rtl while some are hls.
I have included the intermediate models and the build_dataflow_steps.py in the zip file as well as the output received through finn and the expected output. The notebook for running in Pynq is also included. I don't know whether I am doing anything wrong. Also, I took out the last mul node because it is just dividing by 255 to get the output between 0 and 1.  
[check.zip](https://github.com/Xilinx/finn/files/15057930/check.zip)

[Update:]
It works with old layers of v9 but not latest branch of FINN with RTL layers. Since I want more throughput RTL layers are better. Hope this can be solved. 

_Originally posted by @Arbiter-glitch in https://github.com/Xilinx/finn/discussions/1052_

Arbiter-glitch, 2024-04-30:
[Update]:

To find out why the output is different in finn v0.10. I took Finn v0.9 and started to replace each of the old layers with new RTL layers, I checked at each stage. And it was only after replacing the old Thresholding hls layers with RTL layers , the output varied from the expected. So the problem is might be with the thresholding rtl layers.

auphelia, 2024-05-17:
See: Issue https://github.com/Xilinx/finn/issues/1060 and PR https://github.com/Xilinx/finn/pull/1077

closed 2024-05-17 by auphelia