Xilinx/finn #1094: U-Net on PYNQ

phixerino, 2024-06-03:
Hi, I'm trying to run U-Net like segmentation model on PYNQ. I have successfully run the model when I removed skip connections (this is the model after conversion to HW layers: 
[QuantSimpleUnet_512x512x1_1classes_1batch_4W4A_noSkip_upsample_qonnx-hw_layers.zip](https://github.com/user-attachments/files/15537337/QuantSimpleUnet_512x512x1_1classes_1batch_4W4A_noSkip_upsample_qonnx-hw_layers.zip)). But I can't run the model with skip connections. I can synthesize it, but when I try to run it with Python driver, it's stuck on [this line](https://github.com/Xilinx/finn/blob/main/src/finn/qnn-data/templates/driver/driver_base.py#L388). I reduced it only to one skip connection, so that it resembles ResNet block, but the result is the same. For reference, this is the model with one skip connection, after conversion to HW layers: 
[QuantSimpleUnet_512x512x1_1classes_1batch_4W4A_oneSkip_upsample_qonnx-hw_layers.zip](https://github.com/user-attachments/files/15537490/QuantSimpleUnet_512x512x1_1classes_1batch_4W4A_oneSkip_upsample_qonnx-hw_layers.zip)

fpjentzsch, 2024-06-04:
Hi,
did you insert FIFOs of sufficient depth between the graph nodes ? Maybe there is a deadlock at the skip connection.

phixerino, 2024-06-04:
The only FIFO insertion I have is when I call ZynqBuild on [this line](https://github.com/Xilinx/finn/blob/main/src/finn/transformation/fpgadataflow/make_zynq_proj.py#L340). Is that not enough? Should I also call InsertAndSetFIFODepths like in the default build dataflow steps [here](https://github.com/Xilinx/finn/blob/main/src/finn/builder/build_dataflow_steps.py#L573)?

fpjentzsch, 2024-06-04:
Yes, InsertFIFO() will just insert minimally sized FIFOs or no FIFOs (depending on `create_shallow_fifos`) otherwise.
You'll need to set the FIFO depths in one of the ways shown in the "step_set_fifo_depths" step:
- DeriveFIFOSizes() + InsertFIFO()
- InsertAndSetFIFODepths()
- InsertFIFO(create_shallow_fifos=True) + applying manual FIFO depths defined in your folding config json

auphelia, 2024-08-16:
Closing this issue because of inactivity, please feel free to re-open if your problem isn't solved.

closed 2024-08-16 by auphelia