Xilinx/finn #1095: Problem downloading xrt_202220.2.14.354_22.04-amd64-xrt.deb within run-docker.sh

maires1306, 2024-06-03:
After running `./run-docker.sh quicktest` the process gets stuck on step [10/45] trying to download some files.

[image]

Apparently it tries to run `wget https://www.xilinx.com/bin/public/openDownload?filename=xrt_202220.2.14.354_22.04-amd64-xrt.deb`, but it keeps waiting for a response after a HTTP request. I tried to download via browser and it worked fine, but I still don't know where to put the file.

Ba1tu3han, 2024-06-04:
I have encountered this issue for last 3 days. Therefore, I'm using the main brach of v0.10.

judrew, 2024-06-04:
If the download works using your browser, you could use wget with the user-agent set.
e.g. `wget --user-agent="Mozilla/5.0" [...]`

fpjentzsch, 2024-06-04:
This was probably fixed in https://github.com/Xilinx/finn/pull/1092

maires1306, 2024-06-05:
> If the download works using your browser, you could use wget with the user-agent set. e.g. `wget --user-agent="Mozilla/5.0" [...]`

It worked, thanks!

auphelia, 2024-06-05:
Hi all, 
Thanks for (re)-opening the issue. We are working on a solution, if you don't need XRT for your specific use case, try to skip the build by using `dev` and setting `FINN_SKIP_XRT_DOWNLOAD=1`.

auphelia, 2024-06-05:
The xrt download link is back up and you should be able to use `dev` without an issue now.

auphelia, 2024-06-07:
This PR allows to skip the download and instead use a local version of the xrt .deb: https://github.com/Xilinx/finn/pull/1098
Closing this issue for now.

closed 2024-06-07 by auphelia