Xilinx/finn #1097: Issues in getting started on Windows

gian-didom, 2024-06-06:
On Windows, the installation procedure has multiple bugs:

- Since there's a shebang directive `/bin/bash¡ is at the top of `run-docker.sh`, the script gets executed in a MinGW environment - if installed (good)
- Spaces in the username make the script fail since `DOCKER_GNAME=$(id -un)` does not escape spaces
- Most `-v` options when starting the Docker fail due to the fact that MinGW maps `-v /etc/` to a local folder, to which Docker has no access
- When fixing all of the above, docker fails because the folder path is not recognized by Docker as an absolute path (`docker: Error response from daemon: the working directory 'M:/finn' is invalid, it needs to be an absolute path.`)

Can this even be used on Windows?

fpjentzsch, 2024-06-07:
FINN can only be used on Windows via WSL2

auphelia, 2024-06-26:
Hi @gian-didom, I am closing this issue because @fpjentzsch addressed it in his comment. You can find more information in the blog post here: https://github.com/Xilinx/finn/discussions/1084

closed 2024-06-26 by auphelia