Xilinx/finn #1100: out of the range for parameter 'FIFO depth(FIFO_DEPTH)'

Ba1tu3han, 2024-06-08:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [+ ] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [+ ] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [+ ] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
I trained a model with 1W1A and 2W1A quantization then I compiled 1W1A for Pynq-Z2 by FINN however I have encountered a FIFO error for 2W1A. Additionally, I use seperated custom configuration files.

Could you help me about how to set or check FIFO ?

[image]

fpjentzsch, 2024-06-10:
There is a builder config setting to split large FIFOs into smaller ones for this scenario: https://github.com/Xilinx/finn/blob/9f0fad5717ae652eaa324583e957161890964555/src/finn/builder/build_dataflow_config.py#L268

However, I doubt that you should need FIFOs this deep for one of these small models. Do you let FINN determine the FIFO sizing automatically in "step_set_fifo_depths"?

Ba1tu3han, 2024-07-01:
No, I don't know it is automatically or not. I did not change any setting about it. What settings should I do?

Ba1tu3han, 2024-07-03:
In my custom configuration file, I don't determine anything about FIFO. Is it about being a large number of not being power of two? Should I assign them in the config file? I want to everything in compiling step as automatic.

bwintermann, 2024-07-26:
Did you set `split_large_fifos` to true before or after running into the error? If before, then this would mean that the SplitLargeFIFOs() transform fails, since it's supposed to break up too large FIFOs due to exactly the error you encountered. However as @fpjentzsch mentioned, having such large FIFOs shouldn't be needed. Are the layers especially large?

Ba1tu3han, 2024-08-05:
I set _split_large_fifos_ as True and it is solved. I also set _auto_fifo_depths_ as True. Thank you!

closed 2024-08-05 by Ba1tu3han