Xilinx/finn #1122: Issue with launching build of IP: Exception: CreateStitchedIP failed, no wrapper HDL found

1287888420, 2024-07-02:
## Quick summary
I'm new to FINN. I was following the notebooks of end2end_example/cybersecurity. An error was thrown when building up IP when running:
`%%time
build.build_dataflow_cfg(model_file, cfg_stitched_ip)`

## Details
No errors in any previous steps, running on Ubuntu WSL2 with Jupyter Notebook. 
Building dataflow accelerator from /tools/finn/notebooks/end2end_example/cybersecurity/cybsec-mlp-ready.onnx
Intermediate outputs will be generated in /tools/Xilinx/maven/finn0.10/finn/build
Final outputs will be generated in output_ipstitch_ooc_rtlsim
Build log is at output_ipstitch_ooc_rtlsim/build_dataflow.log
Running step: step_qonnx_to_finn [1/19]
Running step: step_tidy_up [2/19]
Running step: step_streamline [3/19]
Running step: step_convert_to_hw [4/19]
Running step: step_create_dataflow_partition [5/19]
Running step: step_specialize_layers [6/19]
Running step: step_target_fps_parallelization [7/19]
Running step: step_apply_folding_config [8/19]
Running step: step_minimize_bit_width [9/19]
Running step: step_generate_estimate_reports [10/19]
Running step: step_hw_codegen [11/19]
Running step: step_hw_ipgen [12/19]
Running step: step_set_fifo_depths [13/19]
ERROR: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:MVAU_hls_1:1.0 
Traceback (most recent call last):
  File "/tools/finn/src/finn/builder/build_dataflow.py", line 158, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/tools/finn/src/finn/builder/build_dataflow_steps.py", line 573, in step_set_fifo_depths
    model = model.transform(
  File "/tools/finn/deps/qonnx/src/qonnx/core/modelwrapper.py", line 140, in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
  File "/tools/finn/src/finn/transformation/fpgadataflow/set_fifo_depths.py", line 325, in apply
    model = model.transform(CreateStitchedIP(self.fpgapart, self.clk_ns))
  File "/tools/finn/deps/qonnx/src/qonnx/core/modelwrapper.py", line 140, in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
  File "/tools/finn/src/finn/transformation/fpgadataflow/create_stitched_ip.py", line 617, in apply
    raise Exception(
Exception: CreateStitchedIP failed, no wrapper HDL found under /tools/Xilinx/maven/finn0.10/finn/build/vivado_stitch_proj_vxj3yg4c/finn_vivado_stitch_proj.srcs/sources_1/bd/finn_design/hdl/finn_design_wrapper.v or /tools/Xilinx/maven/finn0.10/finn/build/vivado_stitch_proj_vxj3yg4c/finn_vivado_stitch_proj.gen/sources_1/bd/finn_design/hdl/finn_design_wrapper.v.
                    Please check logs under the parent directory.
> /tools/finn/src/finn/transformation/fpgadataflow/create_stitched_ip.py(617)apply()
    615                 model.set_metadata_prop("wrapper_filename", wrapper_filename_alt)
    616             else:
--> 617                 raise Exception(
    618                     """CreateStitchedIP failed, no wrapper HDL found under %s or %s.
    619                     Please check logs under the parent directory."""

auphelia, 2024-07-23:
Hi @1287888420 , could you please check the log file in the folder that is linked in the error message? (/tools/Xilinx/maven/finn0.10/finn/build/vivado_stitch_proj_vxj3yg4c/)
That should give you more information about where the error could come from.

auphelia, 2024-08-16:
Closing this issue for now, please feel free to re-open if your problem isn't solved.

closed 2024-08-16 by auphelia