Xilinx/finn #1130: Design built using RTL backend produces incorrect results

LinusJungemann, 2024-07-11:
## Quick summary
When using the RTL backend for synthesizing a simple 2 layer MLP, the rtlsim fails the verification step and the accelerator produces incorrect results. 

## Details
### Steps to Reproduce

1. Clone the FINN repository
2. Checkout the dev branch, with commit hash: f649cdad8e08602ac7d6c353b4c12a3e25694c31
3. Use provided model and configuration: [minimal.tar.gz](https://github.com/user-attachments/files/16173986/minimal.tar.gz)
4. Start the docker container with the command: ./run-docker.sh build_custom minimal (Noctua 2 setup in Paderborn, so instead of docker, singularity is used, but this should not have an impact on the result.)

build_RTLVARIANT.py and build_HLSVARIANT.py can be used to select rtl or hls. The selected file needs to be renamed to build.py and placed in the minimal build folder.
The difference between both build.py files is, that the rtl variant uses the provided specialize_layers.json.

### Expected behavior
Both backends produce the same results.

### Actual behavior
The build using the hls backend produces correct results (+ all verification steps succeed), build using rtl backend produces accelerator with incorrect results (+ rtlsim step fails with incorrect results)

fpjentzsch, 2024-07-11:
I quickly ran your code.

The offending layer seems to be the second MVAU (i.e., it works with all layers but this one switched to RTL). Btw, even your HLS variant ends up with the Thresholding and FIFOs in RTL mode. The second MVAU has the following config:
[image]

Unfortunately, the full output context of the STITCHED_IP_RTLSIM verification is not helpful here because all activation tensors within the "GenericPartition_0" (the actual FINN IP which is surrounded by a MultiThreshold before it (why is this not mapped to the back-end?) and a Mul + Add after it) are reported as [0, ..., 0]. Is this a known issue, @auphelia?

auphelia, 2024-07-12:
Yes, the stitched IP mode can only see the input and the output to the dataflow partition. @hannahxy13 has opened a PR that adds node-by-node rtlsim as a verification step for the builder, in this PR: https://github.com/Xilinx/finn/pull/1087
It is possible to either rebuild the MVAUs in isolation or enabling the verification flow from the PR above could give more insights.

auphelia, 2024-07-16:
This patch fixes your issue: https://github.com/Xilinx/finn/pull/1135

LinusJungemann, 2024-07-18:
Yes, it works now. Thanks!

PR #1135 (2024-07-17): Align datatype extension capabilities between all DSP-based MVU implementations.
Unlike the 8-bit implementations, the DSP-based 4-bit MVU was not configurable to smaller weight or activation bit widths than the nominal 4-bit datatypes. This PR adds this capability and harmonizes the corresponding generic design parameters between the various MVU versions.
This is to fix #1130.
closed 2024-07-18 by LinusJungemann