Xilinx/finn #1133: missing class "InferPool_Batch()"

madnan123, 2024-07-15:
## Quick summary
missing class "InferPool_Batch()"  in convert_to_hls_layers

## Details
1-  "convert_to_hw_layers.py" is incorrectly named as "convert_to_hls_layers" in 
import finn.transformation.fpgadataflow.convert_to_hls_layers as to_hls
### Steps to Reproduce
Add what needs to be done to reproduce the bug. Add code examples where useful
and make sure to include the resulting ONNX files, and the commit hash you are working on.

1. Clone the FINN repository
2. Check the file in src/finn/transformation/fpgadataflow/convert_to_hw_layers
3. There is no class named InferPool_Batch as used in finn_examples resnet50 in custom_steps.py

fpjentzsch, 2024-07-16:
Hi, are you using the dev branch of finn-examples? The main branch is not yet compatible with the latest finn release.

madnan123, 2024-07-17:
As suggested, switching to the development branch resolved the "cycle-free graph violated: partition depends on itself" error I encountered on the main branch. However, I am still facing issues with building the design.
I will attempt to resolve these new issues on my own, but if they persist, I will reach out for further assistance.
Thank you for your help.

Best regards,

closed 2024-07-23 by auphelia